riscv-software-src/riscv-isa-sim #1510: Could anyone tell me whether the address in the commit log is a physical address or a virtual address?

lldb123, 2023-11-20:
Part of the log is as follows:
```
core   0: 0xffffffc000001ec2 (0x00006928) c.ld    a0, 80(a0)
core   0: 1 0xffffffc000001ec2 (0x6928) x10 0x0000000000000001 mem 0xffffffc00041bf10
core   0: 0xffffffc000001ec4 (0x10200073) sret
core   0: 1 0xffffffc000001ec4 (0x10200073) c768_mstatus 0x8000000a000066a2
core   0: 0x000000000003bab8 (0x00007179) c.addi16sp sp, -48
core   0: 0 0x000000000003bab8 (0x7179) x2  0x0000003ffffffa20
core   0: 0x000000000003baba (0x00006285) c.lui   t0, 0x1
core   0: 0 0x000000000003baba (0x6285) x5  0x0000000000001000
core   0: 0x000000000003babc (0x0000ec26) c.sdsp  s1, 24(sp)
core   0: 0 0x000000000003babc (0xec26) mem 0x0000003ffffffa38 0x0000003ffffffb58
core   0: 0x000000000003babe (0x000012c1) c.addi  t0, -16
core   0: 0 0x000000000003babe (0x12c1) x5  0x0000000000000ff0
core   0: 0x000000000003bac0 (0x000074fd) c.lui   s1, 0xfffff
core   0: 0 0x000000000003bac0 (0x74fd) x9  0xfffffffffffff000
core   0: 0x000000000003bac2 (0x00006785) c.lui   a5, 0x1
core   0: 0 0x000000000003bac2 (0x6785) x15 0x0000000000001000
core   0: 0x000000000003bac4 (0x0000f406) c.sdsp  ra, 40(sp)
core   0: 0 0x000000000003bac4 (0xf406) mem 0x0000003ffffffa48 0x000000000001ddb0
core   0: 0x000000000003bac6 (0x0000f022) c.sdsp  s0, 32(sp)
core   0: 0 0x000000000003bac6 (0xf022) mem 0x0000003ffffffa40 0x0000000000000001
core   0: 0x000000000003bac8 (0x0000e84a) c.sdsp  s2, 16(sp)
core   0: 0 0x000000000003bac8 (0xe84a) mem 0x0000003ffffffa30 0x0000003ffffffb68
```
If I understand the meaning of log correctly, for example in this line

`
core 0: 1 0xffffffc000001ec2 (0x6928) x10 0x0000000000000001 mem 0xffffffc00041bf10 
`

`0xffffffc000001ec2` is PC, `0x6928` represents the instruction, `x10 0x0000000000000001` represents the value of the x10 register updated to 0x0000000000000001, and `mem 0xffffffc00041bf10` represents contents of memory with the address 0xffffffc00041bf10, right?

So I would like to ask whether the address behind `mem` is a physical address or a virtual address. Can anyone answer my confusion?

chihminchao, 2023-11-20:
It is virtual address.

lldb123, 2023-11-21:
> It is virtual address.

Thank you for your reply. Currently, I want to obtain the memory access pattern through the commit log. However, due to the large size of the log, it does not seem realistic to manually view the mem access from this log. Do you know if Spike has a way to obtain more direct memory access information?

Saad525, 2023-11-29:
running spike in interactive debug mode might help - https://github.com/riscv-software-src/riscv-isa-sim#interactive-debug-mode