riscv-software-src/riscv-isa-sim #1524: PMP behaviour when granularity is 16 byte

omerguzelelectronicguy, 2023-12-06:
I am using spike wih a flag `--pmpgranularity=16`. I tried to fill pmpaddr0 with `0x00000001ffffffff` while mode is off and pmpaddr0 became `0x00000001fffffffc`

```
core   0: >>>>  pmp_cfg_napot
core   0: 0x00000000800000ca (0x3a0292f3) csrrw   t0, pmpcfg0, t0
core   0: 3 0x00000000800000ca (0x3a0292f3) x5  0x0000000000000000 c928_pmpcfg0 0x000000000000001f
core   0: 0x00000000800000ce (0x3b0022f3) csrr    t0, pmpaddr0
core   0: 3 0x00000000800000ce (0x3b0022f3) x5  0x00000001ffffffff
core   0: >>>>  pmp_cfg_off
core   0: 0x00000000800000d2 (0x3a005073) csrwi   pmpcfg0, 0
core   0: 3 0x00000000800000d2 (0x3a005073) c928_pmpcfg0 0x0000000000000000
core   0: 0x00000000800000d6 (0x3b0022f3) csrr    t0, pmpaddr0
core   0: 3 0x00000000800000d6 (0x3b0022f3) x5  0x00000001fffffffc
```
Here I didn't understand that spike read pmpaddr0 as `0x00000001fffffffc` after changing `pmp0cfg` from off to `napot`. In spec, it is written like below.

> When G ≥ 2 and pmpcfg i .A[1] is set, i.e. the mode is NAPOT, then bits pmpaddr i [G-2:0] read as all ones.

Only the 1-bit LSB should be read as one however, the Spike chosen to read 2-bit LSB.

Which part I am missing here? Does spike behave wrongly?

scottj97, 2023-12-07:
The relevant lines in the spec:

> When G ≥ 2 and pmpcfgi.A[1] is set, i.e. the mode is NAPOT, then bits pmpaddri[G-2:0] read as all ones. When G ≥ 1 and pmpcfgi.A[1] is clear, i.e. the mode is OFF or TOR, then bits pmpaddri[G-1:0] read as all zeros.

You have G=2.

Your first `csrr` has A=NAPOT, so bits `pmpaddr[0:0]` read as all ones.

Your second `csrr` has A=OFF, so bits `pmpaddr[1:0]` read as all zeros.

Apparently, before the trace you pasted above, your program stored a 1 into `pmpaddr[1]`. That's why it's reading back as 1 when A=NAPOT. It's masked to 0 when A=OFF.

omerguzelelectronicguy, 2023-12-07:
Do you mean spec explain as like that?
 When the mode is off and we try to write 11 to those bits, they were, in fact, written but we can't see because of the mode being off. 
Does correct behaviour like that?

scottj97, 2023-12-08:
pmpaddr[0] doesn't need to exist when G=2 because it will always be forced to 1 or 0 depending on MODE. (I mean, there doesn't need to be any physical storage for that bit in the hardware.)

pmpaddr[1] is always writable, even when it's not readable (i.e. when MODE is OFF or TOR).

I'm not sure if the spec requires this writability, but I don't see anything that forbids it. Perhaps this line requires it?

> Although changing pmpcfgi.A[1] affects the value read from pmpaddri, it does not affect the underlying value stored in that register

omerguzelelectronicguy, 2023-12-08:
Ok, thank you for the explanation. I read that part too but I understand differently. I saw some issues concerning those explanations that need clarification on the spec repo (I think too). There is no issue here so I can close it.

closed 2023-12-08 by omerguzelelectronicguy