riscv-software-src/riscv-isa-sim #1533: Why does spike install libfdt.so at build time?

unlsycn, 2023-12-10:
I'm creating a Gentoo ebuild script for spike. And I noticed that spike depends on dtc but at the same time installs libfdt.so, which causes a file collision with dtc. I fixed it simply by substituting $(lib)/spike for install_libs_dir, but I'm still wondering what the difference is between libfdt.so installed by spike and dtc.

aswaterman, 2023-12-11:
cc @jerryz123, who flipped that switch earlier this year.

jerryz123, 2023-12-11:
I thought I flipped the switch to enable construction of external "device plugin libraries", which may need to link against libfdt. I didn't realize that libfdt may be already present on the system from the dtc package.

Now that I'm looking at this more thoroughly, it seems like Spike's fdt code is copied from OpenSBI's, which presumably is copied from the upstream dtc project. This PR: https://github.com/riscv-software-src/riscv-isa-sim/pull/455 pulled the fdt into spike.

It seems Spike should just link against a system libfdt, instead of rebuilding its own.

aswaterman, 2023-12-11:
Unfortunately, libfdt isn't installed on many systems that have dtc, so that change would cause some churn.

My preference would be to avoid making Spike have a new dependence (i.e. continue building libfdt as part of building spike), but stop installing libfdt.so.  For the less-common case that external device plugin libraries need to link against libfdt, can we make only those libraries depend on system libfdt.  Reasonable?

jerryz123, 2023-12-12:
> Unfortunately, libfdt isn't installed on many systems that have dtc, so that change would cause some churn.

Are these systems unable to acquire libfdt easily through a package manager? If it's such an issue, then your suggestion seems reasonable to me.

jerryz123, 2023-12-12:
Ah, its already annoying me. The latest version in ubuntu 20.04 contains a header file bug that breaks compiles.
Let's just remove the spike-installed libfdt then.

aswaterman, 2023-12-12:
APIs... what are they even for?

PR #1535 (2023-12-12): Don't install spike's libfdt to avoid conflicts with system libfdt
Resolves #1533
closed 2023-12-12 by jerryz123