riscv-software-src/riscv-isa-sim #1537: spike run big-endian programs error：trap_load_access_fault

Lebattt, 2023-12-12:
Hi，
   When spike execute a big-endian programs， I got  the following error.
   _terminate called after throwing an instance of 'trap_load_access_fault'._

   However,  there is no this error in little-endian mode.

Thanks in advance.

scottj97, 2023-12-12:
You haven't provided enough info for anyone to diagnose the issue.

What version of Spike are you running? (Binary release? Git hash?)

What target program(s) are you trying to run? How did you compile it?

What Spike command-line did you use for each of these runs?

Lebattt, 2023-12-13:
Sorry，
1.  spike‘s version is 1.1.0 
2.  gcc compilation command contains `-mbig-endian`
3. program is simple initial text and main text， begin address is 0x100000
 
[image]

4. spike command contains  `--isa=rv32imc -l `

Lebattt, 2023-12-13:
@scottj97 thanks for your reply.

I have solved the issue. 
The `cmd.payload() `value in the `syscall_t::handle_syscall()` function is different in big endian or little endian mode. 
I found that the `cmd.payload() `is 0x10000000 in big  endian mode.
However，if tohost add 4byte， the program is successfully executed.

This is because the tohost address load by 8byte align in all endian modes, but my program is 4Byte align. 

Hope this helps.

closed 2023-12-13 by Lebattt