riscv-software-src/riscv-isa-sim #1545: Spike behavior: Mask register overlaps destination

ajayvreddy, 2023-12-20:
I have the following instruction which uses V0 as destination  and also as mask

```
core   0: 0xffffa1000010b0e0 (0xb00540d7) vnsrl.wx v1, v0, a0, v0.t
core   0: 0 0xffffa1000010b0e0 (0xb00540d7) c8_vstart 0x0000000000000000 e8 mf4 l8

```

The destination vector register group for a masked vector instruction cannot overlap the source mask register (v0), unless the destination vector register is being written with a mask value (e.g., compares) or the scalar result of a reduction. These instruction encodings are reserved.

scottj97, 2023-12-22:
It's not clear what the expected behavior vs actual behavior is. My guess:

Expected behavior: illegal instruction trap
Actual behavior: no illegal instruction trap

Is that right?

ajayvreddy, 2023-12-22:
Yes.
I'm expecting it to take a fault (acc to the new spec)

aswaterman, 2023-12-23:
cc @chihminchao

ajayvreddy, 2024-01-02:
Hi @chihminchao  can you please help with this...