riscv-software-src/riscv-isa-sim #1549: What is easiest way to support mounting a virtual disk in Linux?

wsong83, 2023-12-23:
When running Linux using Spike, kernel runs on a ramfs. This has several limitations:
* The drive is mapped to memory, occupying the memory space, and all file read/write through the memory, distorting the memory access pattern.
* Any changes to the ramfs get lost after reboot.
* Available space is limited by the memory (of Spike) and large ramfs image significantly slows down the kernel boot.

I really like to have a virtual disk supported by Spike and can be mounted by the kernel. Ideally, this is a shared directory between host and guest: a directory on the host machine is mapped to the RISC-V Linux as a disk, similar to what the virtiofs supported by QEMU.

I see there are device classes defined both in riscv and fesvr, also there are socket support in riscv for debug? But as I understand, there is currently no support for any kind of virtual disk and no support for virtio (in Spike) as well.

I am not sure I am capable enough for this task, but if I could, I would like to spend some limited time to bring up this function.

Then comes the question, what is the easiest way forward, if I would like to support a shared directory between host and the guest Linux?

I am currently using BBL, do I must switch to OpenSBI, or is this relevant at all?

Is there any good example that I can learn quickly and borrow code?

Many thanks!

wsong83, 2023-12-23:
Is this related? https://github.com/ucb-bar/spike-devices/tree/add-blockdev

jerryz123, 2023-12-30:
^ The spike-devices repo implements a custom block device peripheral device model. That model can be passed a disk image on the host filesystem to present to the target.

To use this block device model, the RISC-V Linux kernel being simulated must be built with a custom block device driver. https://github.com/firesim/iceblk-driver

wsong83, 2023-12-31:
Thanks a lot! This seems very promising and I will have a try.

closed 2024-01-04 by wsong83