riscv-software-src/riscv-isa-sim #1553: The result of fcvtmod_w_d when frs1 is a negative number with large exponent

alextxia13, 2024-01-02:
[image]

I met a question when I run zfa_ext instruction fcvtmod_w_d, the input value is a negative number with large exponent as shown in pic. I think the result of calculation is -0 which is ffffffff80000000 but the spike result is 0. As shown in spike, I saw the code that when sign=1, frac = -frac, but it nor works. Please help to take a look. Thanks!
[image]

nibrunieAtSi5, 2024-01-02:
The expected result for `fcvtmod.w.d` is an integer (32-bit signed integer) not a floating-point. So I don't think we can interpret any result as `-0`. 

`0xea1f7a9391340a41` is a very large (negative) multiple of 2^32 ~2^674, so the integer modulo 2^^32 is expected to be zero. I think spike is right in that case.

alextxia13, 2024-01-02:
Thanks for the reply. But when I run the fcvt.w.d instruction which result is also an integer value not a floating-point with the input value 0xea1f7a9391340a41. The result is 0xffffffff80000000. I think -0 is an available result although it is a 32 bit integer value. 
[image]

nibrunieAtSi5, 2024-01-02:
Beware than in the case of `fcvt.w.d` since there is no modulo, the output is the value which corresponds to invalid inputs:
[image]

alextxia13, 2024-01-02:
I got it. Thanks for your help.

closed 2024-01-02 by alextxia13

nibrunieAtSi5, 2024-01-02:
> I got it. Thanks for your help.

You are welcome.