riscv-software-src/riscv-isa-sim #1557: Question about permission check of a supervisor physical address in a hlvx instruction

viktoryou, 2024-01-05:
As the Priv Spec said
> For the supervisor physical address that results from address translation, the supervisor physical
memory attributes must **grant both execute and read permissions**. (The supervisor physical memory
attributes are the machine’s physical memory attributes as modified by physical memory protection,
Section 3.7, for supervisor level.)

However, when I check the physical memory protection in Spike for this case, only the read permisson is required.
What is expected when the supervisor physical address of a hlvx request is only granted with read permission without execute permission?

viktoryou, 2024-01-10:
Here is some furthur descrption. The content I quoted is added by https://github.com/riscv/riscv-isa-manual/pull/679. To my understanding, the physical memory atttibute modified by physical memory protection should give the requirement of X on the basis of R permission to hlvx instruction. However, it seems the spike hasn't fit this change. From https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/csrs.cc#183, we can see this feature is not added into  `pmpaddr_csr_t::access_ok`. I am not sure if I missed anything. I suspect that maybe a bug in spike and my reproduction testcase convinces me so.

scottj97, 2024-01-10:
For HLVX, Spike checks the PTEs for execute permissions ([here](https://github.com/riscv-software-src/riscv-isa-sim/blob/d6f622e84d7b202f9f1ca23411efca24ea2b5955/riscv/mmu.cc#L442) and [here](https://github.com/riscv-software-src/riscv-isa-sim/blob/d6f622e84d7b202f9f1ca23411efca24ea2b5955/riscv/mmu.cc#L536)).

But I concur that I cannot find a similar check for PMP.

viktoryou, 2024-03-13:
Thanks for the check, Scott. 

Since the issue has been open for a while,  I think it is necessary to mention it again.
@avpatel Could you offer some help on this issue? For a hlvx instruction, what consequence would the lack of execute permission in PMP cause? Is spike simply forgotten to add the check or the check would never added under some consideration?

Any help would be appreciated.

gfavor, 2024-04-04:
Per the current email thread "Permission check when using hlvx" in tech-priv ...

_The HLVX spec says: HLVX cannot override machine-level physical memory protection (PMP), so attempting to read memory that PMP designates as execute-only still results in an access-fault exception._

_This sentence implies that PMP must grant both read and execute permissions, otherwise an access fault results._

As also mentioned in that email thread, an issue should be filed againt riscv-isa-sim (to fix Spike).

closed 2025-03-20 by viktoryou