riscv-software-src/riscv-isa-sim #1571: Compiling spike must require c++17

biglanzi, 2024-01-18:
I want to know if compiling Spike requires an environment with C++17 or higher. My current environment is gcc 4.8.5, which supports only C++11. When I try to compile Spike, it fails because the source code of Spike uses 'optional,' which is a feature of C++17
thanks

chihminchao, 2024-01-18:
C++17 is minimal requirement since spike uses some utilities such as "optional" only avaiable after this version

biglanzi, 2024-01-18:
by the way,my host machine env is centos 7 and gcc 4.8.5
my shell command is 
`cd spke
mkdir build && cd build
../configure --prefix=$RISCV
make -j 8
`

biglanzi, 2024-01-18:
> C++17 is minimal requirement since spike uses some utilities such as "optional" only avaiable after this version

okay,thanks 
I checked the closed issues and found that some of them used c++11, so I was a little confused.
now,I got it, c++17 is minimal requirement after the latest version

closed 2024-01-18 by biglanzi