riscv-software-src/riscv-isa-sim #1572: Please clarify RISCV environment variable in the readme, thanks!

hongbo-miao, 2024-01-18:
I am not very clear about this one:

https://github.com/riscv-software-src/riscv-isa-sim?tab=readme-ov-file#build-steps

[image]

I searched RISC-V tools, however I only found https://github.com/riscv-software-src/riscv-tools which is an archived repo.

Please clarify RISCV environment variable in the readme, thanks! ☺️

jerryz123, 2024-01-18:
That is a bit out of date. Spike can be installed into any path specified by --prefix. An existing riscv toolchain does not need to be already in the install path.

hongbo-miao, 2024-01-29:
Thanks @jerryz123 !
It would be great for someone to help update the documents in future.

jerryz123, 2024-01-29:
A PR would be appreciated :smile: