riscv-software-src/riscv-isa-sim #1574: Going to support Advanced Interrupt Architecture(AIA) ?

ArthurLu, 2024-01-19:
I wonder if spike is going to support Advanced Interrupt Architecture(AIA) in the future? because I am developing related software but have no simulatior supporting AIA. Thanks

aswaterman, 2024-01-19:
I'm not aware of anyone planning to add support for it, but I agree it would be nice to have.

YenHaoChen, 2024-02-06:
The AIA includes three parts, i.e., 1) Smcsrind/Sscsrind extension, 2) Smaia/Ssaia extension, and 3) IMSIC/APLIC device. Spike has an initial implementation for the Smcsrind/Sscsrind extension. I am working on the Smaia/Ssaia extension and aiming to fire a PR next month. I wonder whether Spike would be interested in the IMSIC/APLIC device.

YenHaoChen, 2024-02-23:
I continue developing the Smaia/Ssaia extension and anticipate requiring a few more weeks. Below are some posts that are related to this topic.
https://github.com/riscv/riscv-aia/issues/64
https://github.com/riscv/riscv-aia/issues/66
https://github.com/riscv/riscv-opcodes/pull/232
https://github.com/riscv/riscv-opcodes/pull/233
https://github.com/riscv/riscv-opcodes/pull/234

mmhus, 2025-04-17:
Hi @YenHaoChen is there a branch you are working on? I would perhaps like to see and contribute something for the AIA.