riscv-software-src/riscv-isa-sim #1580: Sync with softfloat 3e

troibe, 2024-01-22:
Spike seems to use a slighty modified version of softfloat **3d** with some additions that are not present in the upstream **3e** version.
Would it be possible to sync them up again?

I'm happy to create the pull requests for this in both repos but I first wanted to make sure I didn't overlook something...
E.g. there is a valid reason the versions are incompatible currently.

On the softloat repo they were suggesting [upstreaming the changes from Spike and then using softloat as a submodule in Spike](https://github.com/ucb-bar/berkeley-softfloat-3/issues/24#issuecomment-1891039292).
Which would ensure the versions stay synced up.

aamartin0000, 2024-01-25:
There is a notion, don't know its timetable or if it's actively being worked on, of replacing the SoftFloat code with Sail (perhaps more accurately, it will be a port SoftFloat to Sail).  Unless there are differences in behavior between this "3d+" and "3e", I would prefer deferring your pull request.

jerryz123, 2024-02-17:
@aswaterman thoughts? 

Personally I think pulling in softfloat as a submodule is cleaner.

> Unless there are differences in behavior between this "3d+" and "3e", I would prefer deferring your pull request.

Updating to the latest softfloat should not get in the way of integrating a SAIL softfloat backend in the future, if that ever becomes desirable.

aswaterman, 2024-02-18:
@jerryz123 I support updating the sources to match SoftFloat-3e, but exactly how is debatable.  We don't want to use the SoftFloat build system; we want to keep using Spike's build system to build SoftFloat.  The simplest way to accomplish that is to overwrite the old sources with the new ones.

Furthermore, submodules make more sense for projects actively being developed.  SoftFloat-3e is a point release and SoftFloat overall is in a state of near-dormancy.

I'm not outright opposed to the submodule approach, but it seems like more trouble than it's worth in this case.