riscv-software-src/riscv-isa-sim #1586: Unexpected Exception in M-Mode Memory Access Outside PMP Range

youzi27, 2024-01-27:
**Type of issue:** bug report

**Impact:** not sure

**Development Phase:** request

**Detailed Description:**

I encountered an unexpected behavior while testing memory access in M-mode using Spike. According to the RISC-V specification,  

> If no PMP entry matches an M-mode access, the access succeeds. 

However, Spike’s behavior seems to contradict the RISC-V specification, which states that in M-mode, the processor should have full access to the physical memory, and accesses outside PMP entries should not be restricted. 

The issue was observed when performing a memory access to an address that lies outside of any configured PMP range in M-mode. Despite the expectation of a successful access based on the RISC-V specification, Spike throws a 'Load access fault' exception 

Could you please clarify if this is the intended behavior in Spike, or if it might be a deviation from the RISC-V specification?

**Environment:**
Spike version: 1.1.1-dev

youzi27, 2024-01-27:
Additional note, the aforementioned situation occurs with the `pmpcfg` set to 0x0f.

youzi27, 2024-01-27:
I've noticed that when the address is far greater than the value defined by PMP, it triggers an exception. Could this be related to the default memory size configured in Spike?

aswaterman, 2024-01-28:
We'd need a complete test case and reproduction instructions to investigate this.  But note the RISC-V spec absolutely does _not_ guarantee that M-mode has unrestricted access to main memory.  Two PMP-related counterexamples come to mind:

- "The matching PMP entry must match all bytes of an access, or the access fails" -- this isn't qualified by privilege mode, and so if an M-mode access spans a PMP region boundary, it'll fail.
- Locked PMPs (L=1) affect M-mode the same as other modes.

aswaterman, 2024-01-28:
I saw you wrote and deleted the comment "I've noticed that when the address is far greater than the value defined by PMP, it triggers an exception. Could this be related to the default memory size configured in Spike?" -- the answer is yes.

youzi27, 2024-04-18:
Thank you for your patient explanation! I will close this issue. Thank you!

closed 2024-04-18 by youzi27