riscv-software-src/riscv-isa-sim #1587: Quick question about cpp compatibility

NazerkeT, 2024-01-28:
Hi everyone, 

I was using Spike for one of my projects and there is a potential need to compile cpp files into riscv and to add new instructions. In this regard, are there any potential limitations of Spike with cpp vs c that I should be aware of? 

Please feel free to let me know where I can refer for information related to this as well. 

Thanks,
Best wishes

jerryz123, 2024-01-28:
Spike is implemented in C++. I'm not sure what limitations w.r.t. language you are thinking of. The build system supports modular development of new ISA extensions.

NazerkeT, 2024-01-29:
Hi @jerryz123 , got it and thanks for your response.

I just thought that Spike supports only C executables, will close the issue now then 👌