riscv-software-src/riscv-isa-sim #1589: Redundant Load and Branch Instructions from HTIF while executing binary

RRavikiran66, 2024-01-31:
When executing the MIbench/telecomm/crc benchmark or any other benchmark in Mibench on the RISC-V ISA simulator (Spike), the execution log shows a repetitive pattern of two identical instructions, specifically load and branch instructions. It is observed that these instructions are originating from HTIF (Host-Target Interface) and not from the ELF file being executed. I have tried to trace it down I found that this pattern begins from mret and continues till sret. So, the Spike enters the machine mode to supervisor mode because of a trap but how can we avoid this? Or it's just the part of Spike.

**Questions:**
Expected Behavior: Is the repetition of identical load and branch instructions from HTIF expected behavior in the Spike simulator when running the MIbench benchmark?

**Mitigation**: How can these redundant load and branch instructions from HTIF be avoided in order to improve the accuracy and efficiency of the simulation? Are there specific configurations or settings that need to be adjusted?

**Environment**:
Host Operating System: Fedora workstation 37
Target Architecture: RISC-V
Benchmark: MIbench/telecomm/crc

**Additional Information**: I have generated the log, below is a small part of the log(I get the same pattern for millions of iterations):

core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core   0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core   0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4
core    0: 0xffffffc000004ea0 (0x00043703) ld      a4, 0(s0)
core    0: 0xffffffc000004ea4 (0xfe070ee3) beqz    a4, pc - 4`

Thank you for your assistance in addressing this issue.

RRavikiran66, 2024-01-31:
Thanks a lot for helping me figure out the issue. Is there a way that I can get the trace with just the PC(IP) that are in the binary that is run on top of Spike Simulator?

sawansib, 2024-01-31:
Hi @scottj97 , is this behavior also expected when Spike is running along with BBL that has full Linux as a payload?

scottj97, 2024-02-01:
> 1)When the INTERLEAVE is set to 1, What I have noticed is it takes lot longer to the run the binary. Is this expected?

Yes, that's expected. You're asking Spike to stop and poll all devices after every single instruction. This is why the interleave was set large in the first place.

2)Is there a way that I can get the trace with just the PC(IP) that are in the binary that is run on top of Spike Simulator?

I believe those PCs are in your binary. (See my next reply.) IIUC the only time Spike would execute instructions not from the target payload is during the first few instructions out of reset, or in debug mode.

scottj97, 2024-02-01:
> Hi @scottj97 , is this behavior also expected when Spike is running along with BBL that has full Linux as a payload?

I don't know anything about BBL but there is HTIF code in OpenSBI, which I assume you're using, and this [spin loop](https://github.com/riscv-software-src/opensbi/blob/bb90a9ebf6d9a2fe7726978d594e82cdbaad7799/lib/utils/sys/htif.c#L108-L109) is the code you see executing over and over.

scottj97, 2024-02-01:
> Thanks again for your quick reply, indeed I'm using OpenSBI.
> 
> As @RRavikiran66 mentioned I also verified and the PC's are not from the binary. I believe that due to certain traps, Spike exits the machine mode by mret and then processes the trap, and then exits the supervision mode by sret. Can this be the case as well?

Remember the PC shown here is a virtual address. I suspect that's why you can't find it in your binary.

Except for external debug, Spike does not have any built-in exception handlers. Any mret or sret that you see in the trace must be coming from your payload. Typically the S-mode handlers (that end with mret) are from OpenSBI, and the S-mode handlers (that end with sret) are from the Linux kernel.

sawansib, 2024-02-02:
> > Thanks again for your quick reply, indeed I'm using OpenSBI.
> > As @RRavikiran66 mentioned I also verified and the PC's are not from the binary. I believe that due to certain traps, Spike exits the machine mode by mret and then processes the trap, and then exits the supervision mode by sret. Can this be the case as well?
> 
> Remember the PC shown here is a virtual address. I suspect that's why you can't find it in your binary.
> 
> Except for external debug, Spike does not have any built-in exception handlers. Any mret or sret that you see in the trace must be coming from your payload. Typically the M-mode handlers (that end with mret) are from OpenSBI, and the S-mode handlers (that end with sret) are from the Linux kernel.

Thank you! I checked the objdump there is no mret or sret instruction. 

I also get this loop when running spike with pk. I have traced the mret and sret instructions to [minit.c](https://github.com/riscv-software-src/riscv-pk/blob/f03685954e825a974e59f278b5c8647440291fc7/machine/minit.c#L252) in riscv-pk. I checked, if you do an objdump of pk you will find 3 mret instructions. The mret exist in call trap_vector, restore_regs and enter_supervisor_mode and i think it is coming from trap_vector call. As if there is no mret in payload it has to come from riscv-pk no?

Sawan

scottj97, 2024-02-02:
> Thank you! I checked the objdump there is no mret or sret instruction.

So your binary includes both OpenSBI and a Linux kernel, yet there's no mret or sret in either of those? 😐 

Here's [six of them](https://github.com/search?q=repo%3Ariscv-software-src%2Fopensbi%20mret&type=code) in OpenSBI.

I think you need to figure out why your objdump is not showing you what you think it's showing you.

> I also get this loop when running spike with pk.

You didn't say anything about pk before. Are you using pk? If so it's part of the binary provided to Spike, just like OpenSBI and the Linux kernel would be.

scottj97, 2024-02-02:
Is there still an open question/issue here? I guess you're asking how to filter the log so it only includes the parts of the execution that you're interested in?

Spike doesn't have any concept of OS, boot loader, SBI, etc. It initializes memory with the blob you provide, and then it runs instructions from that memory, that's all.

Perhaps you want to filter out any exception handlers or kernel/SBI ecalls. You could, in theory, filter the log by privilege mode, so that it prints only when in U-mode. Spike doesn't have any built-in log filtering facility but you could implement this yourself or add it as a post-process.

sawansib, 2024-02-02:
> Is there still an open question/issue here? I guess you're asking how to filter the log so it only includes the parts of the execution that you're interested in?
>
> Spike doesn't have any concept of [...]

Hi @scottj97,

Now it is clear. Thank you so much! If needed, we will add a filter. 

Thank you again!

Sawan

closed 2024-02-02 by scottj97