riscv-software-src/riscv-isa-sim #1599: Could not fetch register "vcsr"; remote failure reply 'E0E'

lumidelta, 2024-02-14:
Good day!
I'm trying to debug Spike via VSCode (using gdb-stub), but registers are not displayed due to this error:
`Could not fetch register "vcsr"; remote failure reply 'E0E'`
There are much more registers that are not displayed in command `info registers csr`. For example:
```
pmpcfg10       0x0	0
pmpcfg11       Could not fetch register "pmpcfg11"; remote failure reply 'E0E'
pmpcfg12       0x0	0
pmpcfg13       Could not fetch register "pmpcfg13"; remote failure reply 'E0E'
pmpcfg14       0x0	0
```
In code I found no difference between this registers, so I don't know what to do.
I'll be grateful for any help!

scottj97, 2024-02-14:
I don't know about `vcsr` but the PMP regs might have something to do with using RV64, which does not have the odd-numbered `pmpcfg` CSRs.

lumidelta, 2024-02-14:
Okay, could you tell me please where I can modify list of displayed registers (which is listed by gdb command -data-list-register-values)?

scottj97, 2024-02-15:
> Okay, could you tell me please where I can modify list of displayed registers (which is listed by gdb command -data-list-register-values)?

Sorry, I don't know anything about Spike's GDB interface. I thought those registers being nonexistent on RV64 might give you some ideas.