riscv-software-src/riscv-isa-sim #1600: Load access fault are not reported for byte-granular accesses

WojciechMula, 2024-02-14:
### Description

I'm purposely trying to access unmapped memory to trigger load fault. Since the memory starts at `0x80000000` I'm accessing address below it, like `0x80000000 - 2 = 0x7ffffffe`. When accessing this address via `lh` instruction, there is an expected exception. When accessing individual bytes using `lb`, it works fine.

Discovered this indirectly, when playing with the indexed vector load instructions.

(Anyway, I feel like I'm missing something obvious.)

### Assembly snippet

```
    li        t0, 0x7ffffffe
    lb        a0, 0(t0)
    lb        a1, 1(t0)
    lh        a2, 0(t0)
```

### Trace

```
core   0: 0x00000000800000ee (0x800002b7) lui     t0, 0x80000
core   0: 3 0x00000000800000ee (0x800002b7) x5  0xffffffff80000000
core   0: 0x00000000800000f2 (0x000032f9) c.addiw t0, -2
core   0: 3 0x00000000800000f2 (0x32f9) x5  0x000000007ffffffe
core   0: 0x00000000800000f4 (0x00028503) lb      a0, 0(t0)
core   0: 3 0x00000000800000f4 (0x00028503) x10 0xffffffffffffffb0 mem 0x000000007ffffffe
core   0: 0x00000000800000f8 (0x00128583) lb      a1, 1(t0)
core   0: 3 0x00000000800000f8 (0x00128583) x11 0x0000000000000000 mem 0x000000007fffffff
core   0: 0x00000000800000fc (0x00029603) lh      a2, 0(t0)
core   0: exception trap_load_access_fault, epc 0x00000000800000fc
core   0:           tval 0x000000007ffffffe
```

### Spike version

It's quite fresh master:

```
commit c5092d5d592c4d5a06759fb1ff1579cc56a4797d (HEAD -> master, origin/master, origin/HEAD)
Merge: 7c890632 6903ff69
Author: Andrew Waterman <andrew@sifive.com>
Date:   Mon Feb 5 11:08:17 2024 -0800

    Merge pull request #1592 from YenHaoChen/patch-1

    Update trigger description in README.md
```

aswaterman, 2024-02-14:
Confirmed.  And quite surprising.  I'm looking into it.

closed 2024-02-14 by aswaterman

WojciechMula, 2024-02-15:
Thank you very much for such a quick fix!

PR #1601 (2024-02-14): Reduce NS16550 address space size to one page
..rather than unbounded, as it used to be.  This led to the rather surprising issue #1600, where a part of the address space assumed to be vacant would allow a subset of accesses.

Resolves #1600