riscv-software-src/riscv-isa-sim #1605: Why using gnu-g++ compiler might generate futex error for a simple hello-world program vs regular elf-g++ compiler?

NazerkeT, 2024-02-18:
Hi there, 

I am having this weird error despite double checking files and outputs multiple times. I wanted to run more complex file with `riscv64-unknown-linux-gnu-g++` compiler, however it kept outputting `The futex facility returned an unexpected error code` error. Accordingly, I decided to run simple `hello, world` program with the same compiler to check the results, and weirdly it gave me the same error. In comparison, after compiling the program with `riscv64-unknown-elf-g++` compiler, hello-world launches as expected. When I run `hello, world` program in c, both versions of gcc compiler worked as expected as well. 

My command was as simple as: `riscv64-unknown-linux-gnu-g++ -static -o hello hello.cpp` and `spike pk hello', the same goes for every other compiler versions.

Do you have by chance any ideas on why it might be happening? Is there any recent bug in gcc/g++?

Thanks,
Best wishes

aswaterman, 2024-02-19:
`pk` doesn't support the futex system calls.  glibc used to work correctly without them, but these days, it assumes a newer version of the Linux kernel, and so it assumes the futex calls exist.

jerryz123, 2024-02-19:
Another alternative is to hack `sys_futex` as a no-op into syscalls.c in pk. This should generally not cause problems, as pk doesn't support threading anyways.

NazerkeT, 2024-02-20:
Oh, I see, got it, thanks @aswaterman and @jerryz123 !

closed 2024-02-20 by NazerkeT