riscv-software-src/riscv-isa-sim #1608: Override pc address

lovisXII, 2024-02-18:
Hello, I'm trying to override the boot address of spike using --pc options. First of all it's not specified anywhere the type of value which is expected by after reading the doc it seems to use strtoull and convert the value to an int, so we need to add 0x for hexadecimal values.

Then when the programm boot it seems it's not finding data at the boot address I have provided. My programm is supposed to start at : 
`bfc00000:	c0400e17          	auipc	t3,0xc0400`
But when I check inside spike I have :
```
(spike) mem 0 0xbfc00000
0x00000000
(spike) pc 0
0xbfc00000
```
So how is it working and also is there a doc somewhere, because the readme is not very detailled

jerryz123, 2024-02-19:
Spike specifies the symbol table in the ELF file to figure out how to write the target binary into its main memory region. 
You may want to check that your ELF has that address is in a PT_LOAD segment.

Btw, Spike figures out the start PC from the ELF by looking at the `e_entry` field in the header. You don't have to use `--pc` to override the start pc if `e_entry` already points to the address you want

lovisXII, 2024-02-20:
Thanks for the answer I did not know about the PT_LOAD
Yes but I use the programm for other purpose that just spike and I don't want to modify all my sections.
btw anygood reference for PT_LOAD because I tried to modify one of my segment to make it PT_LOAD but it doesnt seems to work

lovisXII, 2024-02-20:
I've finally succeed to force my sections such as :
```
seg_reset   = 0x80001000;
seg_user    = 0x1001000;
ENTRY(_reset)

PHDRS
{
   user PT_LOAD FILEHDR PHDRS;
   system PT_LOAD FILEHDR PHDRS AT(seg_reset - 0x1000);
}

SECTIONS
{
   . = seg_reset;
   .reset : { *(.reset) } : system
   . = ALIGN(0x1000);
   .kernel : { *(.kernel) } : system
   . = ALIGN(0x1000);
   .exit   : { *(.exit) } : system

   . = seg_user;
   .text         : { *(.text) } : user
   . = ALIGN(0x1000);
   .tohost       : { *(.tohost) }
   . = ALIGN(0x1000);
   .data         : { *(.data) }
   .data.string  : { *(.data.string)}
   .bss          : { *(.bss) }
   _end = .;

}

```
But when I try to launch spike on the executable I got Memory address 0x1000000 is invalid. I tried different values but with all the combinaison I tried I get the same error