riscv-software-src/riscv-isa-sim #1609: How to describe instruction behavior in file riscv/insns/<my instruction>.h?

niuzhi1221, 2024-02-19:
Is there a document for this?

aswaterman, 2024-02-19:
No, it’s basically RTFC.

niuzhi1221, 2024-02-19:
> No, it’s basically RTFC.

Hello! I designed an R-type instruction with a string in rs1 and a character in rs2. If I want to replace the first character in the string with the character in rs2 and write the modified string into rd, how should I describe this instruction behavior? I sincerely hope to receive your guidance.

scottj97, 2024-02-21:
Have you looked at how some of the other simple arithmetic opcodes are written? They are trivial.

You also need to add it to encoding.h (from the riscv-opcodes project) and the riscv.mk.in Makefile, and teach disasm.cc how to disassemble it.

If you want to use your new opcode in an assembly language program, you'll need to teach the assembler about it, but that's outside the scope of Spike.

WojciechMula, 2024-12-13:
Sorry for commenting the closed issue, but I think this maybe helpful hint. The specification on the ISA level is given by https://github.com/riscv/sail-riscv/tree/master; Spike provides C++ interpretation of that spec.