riscv-software-src/riscv-isa-sim #1614: Is there any upper memory bound that spike can allocate?

NazerkeT, 2024-02-29:
Hi there, 

I was recently running OpenSBI + Linux on Spike and root launches as expected with regular small rootfs payload. However, when I rerun simulation with 3.6GB of data added to the rootfs overlay, I hit kernel panic at the very last moment with it not being able to mount rootfs. I tried to modify maximum memory amount passed with -mN as well, but it did not work. Is there anything I need to know more to be able to diagnose this problem better?

Particularly, the following are my logs.  

`
OpenSBI v1.2
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : ucbbar,spike-bare
Platform Features         : medeleg
Platform HART Count       : 1
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 10000000Hz
Platform Console Device   : htif
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : htif
Platform Shutdown Device  : htif
Firmware Base             : 0x80000000
Firmware Size             : 212 KB
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*
Domain0 Region00          : 0x0000000002080000-0x00000000020bffff (I)
Domain0 Region01          : 0x0000000080000000-0x000000008003ffff ()
Domain0 Region02          : 0x0000000002000000-0x000000000207ffff (I)
Domain0 Region03          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000082200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.11
Boot HART Base ISA        : rv64imafdc
Boot HART ISA Extensions  : none
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 54
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109
[    0.000000] Linux version 6.1.14 (nazerket@nazerket24.c.googlers.com) (riscv64-buildroot-linux-gnu-gcc.br_real (Buildroot 2023.11-1184-gd8ea853b8c-dirty) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Tue Feb 20 03:37:46 UTC 2024
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Machine model: ucbbar,spike-bare
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000047fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000047fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x000000047fffffff]
[    0.000000] SBI specification v1.0 detected
[    0.000000] SBI implementation ID=0x1 Version=0x10002
[    0.000000] SBI TIME extension detected
[... 118 lines cut ...]
[    0.169540] In-situ OAM (IOAM) with IPv6
[    0.169815] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    0.170570] NET: Registered PF_PACKET protocol family
[    0.170990] 9pnet: Installing 9P2000 support
[    0.171410] Key type dns_resolver registered
[    0.178600] debug_vm_pgtable: [debug_vm_pgtable         ]: Validating architecture page table helpers
[    0.181025] /dev/root: Can't open blockdev
[    0.181275] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    0.181720] Please append a correct "root=" boot option; here are the available partitions:
[    0.182215] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    0.182705] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.1.14 #1
[    0.183060] Hardware name: ucbbar,spike-bare (DT)
[    0.183345] Call Trace:
[    0.183495] [<ffffffff80005460>] dump_backtrace+0x1c/0x24
[    0.183825] [<ffffffff8073b12e>] show_stack+0x2c/0x38
[    0.184140] [<ffffffff807451e2>] dump_stack_lvl+0x3c/0x54
[    0.184470] [<ffffffff8074520e>] dump_stack+0x14/0x1c
[    0.184780] [<ffffffff8073b3dc>] panic+0x106/0x29e
[    0.185075] [<ffffffff808014c6>] mount_block_root+0x170/0x222
[    0.185425] [<ffffffff80801768>] mount_root+0x1f0/0x220
[    0.185745] [<ffffffff808018da>] prepare_namespace+0x142/0x184
[    0.186100] [<ffffffff808010ae>] kernel_init_freeable+0x23e/0x260
[    0.186470] [<ffffffff807457be>] kernel_init+0x1e/0x10a
[    0.186790] [<ffffffff800034a4>] ret_from_exception+0x0/0x16
[    0.187135] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0) ]-`

aswaterman, 2024-02-29:
The `-m` flag sets how much memory the target machine has.  Beyond that, there is no limit (beyond the host machine running out of memory, which of course would not manifest in this manner).

NazerkeT, 2024-02-29:
I see, nevertheless, can using initrd somehow overwrite spike's memory limits? Because interestingly, if I remove initrd flag, I can dump dts that sets memory to the asked amount, unless otherwise I get following error message:

 

> Error: <stdin>:9.27-38 Value out of range for 32-bit array element
Error: <stdin>:10.25-36 Value out of range for 32-bit array element
FATAL ERROR: Syntax error parsing input tree
Child dtb process failed

And command and script that I am running for spike are the following:

`./board/spike/riscv64/start.sh -m8192`

> #! /bin/sh

> SCRIPT_DIR="$(dirname "$0")"
> BR_BASEDIR="$(readlink -e "${SCRIPT_DIR}/../../..")"

> export PATH="${BR_BASEDIR}/output/host/usr/bin:$PATH"
> : "${SPIKE:=spike}"

>exec "${SPIKE}" \
>    --initrd "${BR_BASEDIR}"/output/images/rootfs.cpio \
>    "${@}" \
>    "${BR_BASEDIR}"/output/images/fw_payload.elf

mickpiotrowski, 2025-03-25:
@NazerkeT Any success in finding out what's going on? When I've tried to specify over default -m2048 I've encountered the same problem.
With default memory I cannot load ramdisk.