riscv-software-src/riscv-isa-sim #1621: Clarification needed for Stopcount and Stoptime bits in DCSR ( in spike )

abhi-valtrix, 2024-03-11:
I have tset stopcount and stoptime in spike using abstract command to write dcsr  - 

riscv dmi_write 0x4 0x400
riscv dmi_write 0x17 0x3307b0

I checked the minstret and time csrs before and after setting them,  they seem to be incrementing.

Are these fields not yet functional in spike ? or am I missing something ?

abhi-valtrix, 2024-03-22:
Hi , 

Any inputs on this ? 

Thanks

YenHaoChen, 2024-07-31:
Unfortunately, Spike does not implement dcsr.stopcount and dcsr.stoptime. In other words, the dcsr.stopcount and dcsr.stoptime are read-only 0 in Spike.

I can provide a PR for the dcsr.stopcount and dcsr.stoptime if desired.

aswaterman, 2024-07-31:
`stoptime` is not appropriate for multicore systems, so we should not implement in Spike.  (Theoretically, we could implement it only for single-core systems, but that seems like a weird special case that I would recommend against.)

If we want to implement `stopcycle`, I would like to do it in a way that does not add any instructions to the main simulation loop.  In particular, if we could accomplish this by only adding an if-statement to the following line, I'd be OK with it.  But I don't think it's high priority since it's optional.  https://github.com/riscv-software-src/riscv-isa-sim/blob/4703ad98bf4c247a0841a6d7254357b14a97ff29/riscv/execute.cc#L350