riscv-software-src/riscv-isa-sim #1623: Instruction to instruction inconsistency in writing vstart on instructions that require vstart==0

rbuchner-aril, 2024-03-11:
Looking though `riscv/insns`, several instructions explicitly require that `vstart==0`

```console
$ grep -rn 'require(P.VU.vstart->read() == 0);'
vmsbf_m.h:4:require(P.VU.vstart->read() == 0);
vcpop_m.h:6:require(P.VU.vstart->read() == 0);
viota_m.h:8:require(P.VU.vstart->read() == 0);
vmsof_m.h:4:require(P.VU.vstart->read() == 0);
vmsif_m.h:4:require(P.VU.vstart->read() == 0);
vcompress_vm.h:2:require(P.VU.vstart->read() == 0);
vfirst_m.h:6:require(P.VU.vstart->read() == 0);
```

However, I noticed that only some of these instructions still write 'vstart' (`P.VU.vstart->write(0);`) at the end of the instruction. For instance, `vfirst.m` still writes `vstart`, but `vmsif.m` does not.

Either choice behaves correctly because `vstart` is already 0, but it is unclear to me why some write and some don't. I was hoping for some clarification for why this difference exists since I can't find anything in the specification on the subject; as far as I can tell, it is an artifact of the development process. I can open a MR to reconcile these differences if that would be appropriate.

aswaterman, 2024-03-12:
We had a discussion on this topic on #1570 and concluded that, when there's a `require(vstart == 0)`, we should delete the explicit write of `vstart`.  In #1573, some, but not all, of these were deleted.  If you want to finish the job, go for it!

rbuchner-aril, 2024-03-12:
Thanks for the response. Looks like I was slightly behind the most recent `master` when making this report; I think only `vmcompress.vm` has this issue still.

However, it utilizes `VI_LOOP_END` macro, which contains
```
#define VI_LOOP_END \
  } \
  P.VU.vstart->write(0);
```

Just wanted to check to see that you are fine removing this macro and replacing it with a single `}`. Another option would be to create a subset of this macro but that seems like overkill for this one case, though it could be better in the long run in case VI_LOOP_END ever needs updated.

`v_ext_macros.h`
```c
#define VI_LOOP_END_BASE \
  }

#define VI_LOOP_END \
  VI_LOOP_END_BASE \
  P.VU.vstart->write(0);
```

`VI_LOOP_END_BASE` would only be used in `vcompress_vm.h`

Thoughts?

aswaterman, 2024-03-12:
The second option seems cleaner to me since it keeps the implementation of vcompress.vm more symmetric.  OK with you, too, @jerryz123?

jerryz123, 2024-03-12:
Yes, the second approach seems more future proof for future extensions which will pattern-match off these definitions

aswaterman, 2024-03-12:
Resolved by #1624

closed 2024-03-12 by aswaterman