riscv-software-src/riscv-isa-sim #1633: Label and register recognition in interactive debug mode

dkaszews, 2024-03-27:
I like to use spike's interactive debugger for quick testing, but it is inconvenient to always have to provide memory addresses manually.

## Issue
If I want to read memory at stack pointer, I first need to check the stack pointer, then copy the value and paste it into a separate command:

```shell
(spike) reg 0 sp
0x0000000080001234
(spike) mem 0 0x0000000080001234
0x00000000deadbeef
```

Even worse is when trying to run to a function, as I need to use `objdump` in a separate window to find the address, despite spike clearly being aware of it as it prints labels during noisy execution:

```shell
$ riscv-unknown-elf-objdump -T hello | grep ' foo$' | awk '{ print $1 }'
00000000080005000
(spike) until pc 0x00000000080005000
(spike) pc 0
00000000080005000
```

## Proposal
Allow interactive debugger to resolve addresses from registers and labels, even better with offsets. Syntax could be any, some or all of:

```
mem 0 reg
mem 0 reg+10
mem 0 (reg)
mem 0 10(reg)
mem 0 @reg 
mem 0 @reg+10

until pc 0 label
until pc 0 label+10
until pc 0 :label
until pc 0 :label+10
until pc 0 label:
until pc 0 label:10
```

jerryz123, 2024-05-13:
This could be useful, but would require substantial rearchitecting of the `interactive` REPL. That code is already messy

dkaszews, 2024-05-13:
Labels may be a stretch, but looking at 
https://github.com/riscv-software-src/riscv-isa-sim/blob/c81d8e73daf1875d79a54f28f67df150fba0e44c/riscv/interactive.cc#L636-L652

and

https://github.com/riscv-software-src/riscv-isa-sim/blob/c81d8e73daf1875d79a54f28f67df150fba0e44c/riscv/interactive.cc#L589-L590

registers should be easy enough.