riscv-software-src/riscv-isa-sim #1638: illegal instruction trap at vlm.v instr

syedarafia13, 2024-04-04:
Hi, I'm facing some problems with the vadd instruction. 
For example, when I run the test for vadd (specifically, vadd.vv_mf8_rm0_int64_0_0_vl4, which means fractional lmul=1/8, sew=64, and vlen=4) on Spike, it gives an illegal instruction trap at the vml.v instruction. For this specific combination of fractional lmul (mf8), sew (64), and vlen (4), Spike hangs. Is this combination valid or not? Or is there an issue with Spike for this combination?
I 've also attached the screenshot of the error given below.
[image]
Please help to take a look, Thanks!

aswaterman, 2024-04-04:
There are two problems here.  First, SEW=64 LMUL=1/8 isn't guaranteed to be supported.  In general, you can only assume that LMUL can go as low as (SEW / ELEN), or 1 in this case.  You can only use LMUL=1/8 for SEW=8 and ELEN=64.

The second problem is that `vlm` uses EEW=8, which means its EMUL (== LMUL / SEW * EEW) is 1/64, which is also illegal.

syedarafia13, 2024-04-04:
Make sense to me, but then I have 2 more questions:
1) Why **vadd.vv_mf8_rm0_int64_1_0_vl4** is passing (same config but unmasked)?
2) Then, what should be the legal SEWs and EEWs for LMUL=1/8?

aswaterman, 2024-04-04:
As I said, SEW=64 LMUL=1/8 isn't _guaranteed_ to be support it.  A given implementation might support it.  My guess is that `vadd.vv_mf8_rm0_int64_1_0_vl4` is not a portable test program; it'll work on some implementations and not on others.

syedarafia13, 2024-04-04:
And what about this test case (vadd.vv_mf8_rm0_int32_0_0_vl8). when lmul=1/8, sew=32 and vlen=8. Is this test scenario is legal or not? Or how we calculate the legal test cases on the basis of lmul and sew?

WojciechMula, 2024-04-05:
You can always check `vill` field of `vtype` register after `vsetvli`.