riscv-software-src/riscv-isa-sim #1643: Supporting variable XLEN

aswaterman, 2024-04-11:
No one has been beating down the door requesting changeable XLEN, but since it'll probably happen eventually, I wanted to have a place to take notes on what changes need to be made.  (cc @jerryz123 @scottj97)

- In XLEN=32 mode, Spike currently assumes that the x-registers' contents are all canonical sign-extended.  Under that premise, instructions are guaranteed to always produce canonical sign-extended values.  However, the spec says that the implementation should ignore the upper XLEN-32 bits, rather than assuming they are canonical (https://github.com/riscv/riscv-isa-manual/blob/217b3f8cdbd94583f4933bb7c7b35bae3e70f74d/src/machine.adoc#L500-L505).  HOWEVER, I think the following clause in the spec allows an alternative implementation, where we canonicalize the MSBs whenever XLEN changes.  That would be much simpler to implement.  https://github.com/riscv/riscv-isa-manual/blob/217b3f8cdbd94583f4933bb7c7b35bae3e70f74d/src/machine.adoc#L232-L233

aamartin0000, 2024-04-19:
Question about this: assuming you are on an RV64 machine (== native XLEN=64), if you change to XLEN=32, are you expecting it to produce results as if it was an RV32 machine, or will there be discrepancies?  There are some RV64-only instructions-- will they be treated as illegal?  There are some RV32-only instructions (e.g. scalar crypto), do these become legal?

The scenario I'm thinking of is being able to run a binary compiled for RV32, on an RV64 machine.

aswaterman, 2024-04-19:
An RV64-native machine with XLEN set to 32 should behave just the same as a native RV32 machine.  Now of course the machine could have some non-conforming extensions (e.g. the RV64-only ADDW instruction might be present when XLEN=32, as a non-conforming extension), but then the implementation would not be able to claim that it implements the Ssstrict extension.