riscv-software-src/riscv-isa-sim #1645: [BUG] Spike incorrectly executes V extension instructions when vsstatus.vs is off and V=1.

youzi27, 2024-04-17:
### **Bug Description**
Hi,

I found Spike can execute vector extension instructions when vsstatus.vs is 0, which violates the definition of the RISC-V V extension specification: 

> When V=1, both vsstatus.VS and mstatus.VS are in effect: attempts to execute any vector instruction, or to access the vector CSRs, raise an illegal-instruction exception when either field is set to Off.
 
**The log is as follows:**

```
(spike) reg 0 mstatus
0x0000000a00102280
(spike) reg 0 vsstatus
0x0000000200000000
(spike) 
core   0: 0x000000008000006e (0x01b2f2d7) vsetvli t0, t0, e64, m8, tu, mu
core   0: 3 0x000000008000006e (0x01b2f2d7) x5  0x000000000000000f c8_vstart 0x0000000000000000 c3104_vl 0x000000000000000f c3105_vtype 0x000000000000001b c768_mstatus 0x8000000a00102680
```

If I have any misunderstandings about this, please let me know. Thank you very much!

aswaterman, 2024-04-17:
Are you certain that V=1 in this case?  I looked at the code, and at least by inspection, it does appear to correctly check vsstatus.VS when V=1.

Assigning to @chihminchao

youzi27, 2024-04-18:
Although I expected it to operate in V mode, after debugging, I discovered that the aforementioned issue occurred in the exception handler before entering V mode. According to the RISC-V priv specification : 

>When V=0, vsstatus does not directly affect the behavior of the machine

Therefore, it complies with the specifications, and I will close this issue. I apologize for the inconvenience. 

Thank you!

closed 2024-04-18 by youzi27