riscv-software-src/riscv-isa-sim #1647: For the vfmv.v.f, fmv.w.x, fmv.x.w instructions, should spike consistently ignore the higher 32 bits when processing single-precision floating-point numbers?

Shanchuang666, 2024-04-18:
We found that when spike executed the following floating-point transfer class instructions, the register high bit processing strategy was inconsistent, is it better to keep the strategy unified?

①vfmv.v.f : When sew=32, spike will **check whether the high 32 bits of the floating-point register are all ones**. If not, it will become the canonical NaN (0x7fc00000) and then operate.
②fmv.w. x: Writes 32-bit data to the floating-point register, **ignoring high 32-bit data in the source register**. The high 32 bits of the destination floating-point register write all ones.
③fmv.x.w：The low 32-bit data in the floating-point register is read, **the high 32-bit data is ignored,** and no modification is made to the low data during the move.

aswaterman, 2024-04-18:
Spike is correctly implementing what the ISA specification requires in these cases.

Shanchuang666, 2024-04-19:
> Spike is correctly implementing what the ISA specification requires in these cases.

Thank you very much for your reply. 
We understand that spike complies with the ISA specification, but we want our microarchitecture implementation to have both scalar and vector floating-point move instructions ignore the high 32 bits of the register (that is, neither check whether the high 32 bits of the register are all ones). Therefore, we modify the spike of the vfmv.v.f instruction on our own to ignore whether the register high 32 bits is correct when the sew is 32. 
Thank you very much for your reconfirmation.

aswaterman, 2024-04-19:
Of course you are entitled to build a non-conforming implementation!

closed 2024-04-19 by aswaterman