riscv-software-src/riscv-isa-sim #1651: mcontrol6.timing bit is set

JAYANTH-IITM, 2024-04-24:
While trying to implement type 6 trigger (mcontrol6) with load bit enabled according to the debug spec 1.0 but spike doesnt seem to support (still using 0.13 version) . tinfo.version == 0 , implies implementation of older debug spec.This issue is persistent also after updating to latest version of spike , openocd  and riscv-tests.After the checking the log (attached below) , dmstaus.version == 2 , which according to latest spec , means version 0.13. Expected to implement in version 1.0 debug spec ( dmstatus.version == 3). PFA , log  after updating spike and openocd .

[20240424-122230-spike32-TriggerDmode.log](https://github.com/riscv-software-src/riscv-isa-sim/files/15090047/20240424-122230-spike32-TriggerDmode.log)

JAYANTH-IITM, 2024-04-30:
@aswaterman , please guide here .

aswaterman, 2024-04-30:
Sorry, I don't maintain debug support in Spike.  Perhaps @timsifive can comment.

JAYANTH-IITM, 2024-04-30:
Thanks for the referring .