riscv-software-src/riscv-isa-sim #1656: How can I read content from a file and use it as input for an ELF file

Thirty-creed, 2024-04-28:
I have written a test program as follows:
~~~c
#include <stdio.h>

int main() {
    int a = 0;
    scanf("%d", &a);
    printf("output:%d\n", a);
    return 0;
}
~~~
I compiled it using the command`clang -target riscv64-linux-gnu -static test.c`  and executed the `spike /riscv64-linux-gnu/bin/pk a.out` command. Then, I manually entered 100, and it outputted the correct result 100, just like this:
~~~bash
$clang -target riscv64-linux-gnu -static test.c
$spike /riscv64-linux-gnu/bin/pk a.out
100
output:100
~~~
But when I used another command `spike /riscv64-linux-gnu/bin/pk a.out < input.txt`(the content of input.txt is 100), it didn't output the expected 100 but instead 0, just like this:
~~~bash
$spike /riscv64-linux-gnu/bin/pk a.out < input.txt
output:0
~~~
Is there an error in my usage? What should I do?

jerryz123, 2024-05-13:
fd-0 (stdin) is redirected to the ns16550 UART device model, so piping data into spike via stdin will not work.

A `fopen` in your target program will work.

closed 2024-05-13 by jerryz123

Thirty-creed, 2024-05-14:
Thanks for your reply.
I'm sorry that my question misled you. I don't want to change my test program, in other words, I don't want to use `fopen`. How can I pass the contents of a file to my program? Is it possible to achieve this by modifying the shell command（`spike /riscv64-linux-gnu/bin/pk a.out < input.txt`）?

Thirty-creed, 2024-05-15:
OK. Thank you again for your reply.