riscv-software-src/riscv-isa-sim #1664: satp written with an unsupported MODE should have no effect

poemonsense, 2024-05-08:
According to the RISC-V spec, when `satp` is written with an supported MODE, no fields in `satp` are modified.

> Implementations are not required to support all MODE settings, and if `satp` is written with an
> unsupported MODE, the entire write has no effect; no fields in `satp` are modified.

However, Spike seems to allow asid/ppn to be changed when the MODE is invalid (`satp_valid` decides only whether the mode_mask takes effect). This behavior seems inconsistent with the ISA manual.

Source: https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/csrs.cc#L980-L983

```c++
reg_t base_atp_csr_t::compute_new_satp(reg_t val) const noexcept {
  reg_t rv64_ppn_mask = (reg_t(1) << (MAX_PADDR_BITS - PGSHIFT)) - 1;

  reg_t mode_mask = proc->get_xlen() == 32 ? SATP32_MODE : SATP64_MODE;
  reg_t asid_mask_if_enabled = proc->get_xlen() == 32 ? SATP32_ASID : SATP64_ASID;
  reg_t asid_mask = proc->supports_impl(IMPL_MMU_ASID) ? asid_mask_if_enabled : 0;
  reg_t ppn_mask = proc->get_xlen() == 32 ? SATP32_PPN : SATP64_PPN & rv64_ppn_mask;

  // Here only mode_mask is decided by satp_valid, while asid and ppn always take effect.
  reg_t new_mask = (satp_valid(val) ? mode_mask : 0) | asid_mask | ppn_mask;
  reg_t old_mask = satp_valid(val) ? 0 : mode_mask;

  return (new_mask & val) | (old_mask & read());
}
```

Look like this bug has existed since the first time WARL feature is added to the satp register (https://github.com/riscv-software-src/riscv-isa-sim/commit/bed716c601f5140afd595ea60f20d9eddab02032) and has gone through several fixes (such as https://github.com/riscv-software-src/riscv-isa-sim/commit/ce170a71eb1ab5296b547fdeafe40c9448e9cdda).

poemonsense, 2024-05-08:
If confirmed a bug, I'm happy to file a patch for fix.

poemonsense, 2024-05-08:
Given the maintainers of rocket-chip and Spike (and riscv-isa-manual) may overlap @aswaterman , I would like to add a few comments on Rocket as well. It seems to have a similar issue where in M mode, asid and ppn will be updated even if mode is invalid. If confirmed a bug, I can assist to fix rocket-chip as well.

https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/rocket/CSR.scala#L1414-L1424
```scala
      when (decoded_addr(CSRs.vsatp)) {
        val new_vsatp = wdata.asTypeOf(new PTBR())
        val mode_ok = new_vsatp.mode.isOneOf(satp_valid_modes.map(_.U))
        when (mode_ok) {
          reg_vsatp.mode := new_vsatp.mode & satp_valid_modes.reduce(_|_).U
        }

        // Here when !reg_mstatus.v, ppn and asid may be updated even if mode is not ok
        when (mode_ok || !reg_mstatus.v) {
          reg_vsatp.ppn := new_vsatp.ppn(vpnBits.min(new_vsatp.ppn.getWidth)-1,0)
          if (asIdBits > 0) reg_vsatp.asid := new_vsatp.asid(asIdBits-1,0)
        }
      }
```

aswaterman, 2024-05-08:
My reading of the rule is that does not apply when V=0 and there’s an explicit write to vsatp. So I don’t think there’s a bug in rocket.

poemonsense, 2024-05-08:
> My reading of the rule is that does not apply when V=0 and there’s an explicit write to vsatp. So I don’t think there’s a bug in rocket.

I just misread v to be the prv in my original comment (and thought `!mstatus.v` was for not in M mode).

But if I'm reading the mstatus.v correctly now, it is asserted if the hart is in virtualized mode. `!reg_mstatus.v` means we are in the normal mode, which is always true if we don't have the hypervisor extension. Thus, that condition becomes always true if Hypervisor extension is disabled. That is, ppn and and asid will always be updated even if the mode of the new_satp is invalid. This behavior seems the same as Spike.

Since you are not mentioning whether Spike has a bug, could you please explain a bit more on this part? Thanks very much for your time. For simplicity, both answers to following questions are "does" for me, because I assume if the `mode` field of written value for `satp` is illegal, the full `satp` register remains unchanged. 

1) Spike (does / does not) have a bug

2) Rocket (does / does not) have the same behavior as Spike when hypervisor is disabled

poemonsense, 2024-05-08:
> My reading of the rule is that does not apply when V=0 and there’s an explicit write to vsatp. So I don’t think there’s a bug in rocket.

Sorry the code is for `reg_vsatp` instead of `reg_satp`. I'm reading the wrong rocket code.

This one is correct. https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/rocket/CSR.scala#L1340-L1349

```scala
      when (decoded_addr(CSRs.satp)) {
        if (usingVM) {
          val new_satp = wdata.asTypeOf(new PTBR())
          when (new_satp.mode.isOneOf(satp_valid_modes.map(_.U))) {
            reg_satp.mode := new_satp.mode & satp_valid_modes.reduce(_|_).U
            reg_satp.ppn := new_satp.ppn(ppnBits-1,0)
            if (asIdBits > 0) reg_satp.asid := new_satp.asid(asIdBits-1,0)
          }
        }
      }
```

The answer to the second question is "does not". Rocket will not update ppn/asid when the written satp mode is illegal.

This behavior seems different with what I see for Spike.

poemonsense, 2024-05-08:
> This behavior seems different with what I see for Spike.

My fault. Spike checks again the valid mode here in `unlogged_write` (https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/csrs.cc#L1015-L1019). 

```c++
bool virtualized_satp_csr_t::unlogged_write(const reg_t val) noexcept {
  // If unsupported Mode field: no change to contents
  const reg_t newval = orig_satp->satp_valid(val) ? val : read();
  return virtualized_csr_t::unlogged_write(newval);
}
```

Now the first answer is also "does not" now. Thanks for your time. I'm closing this issue now.

closed 2024-05-08 by poemonsense

aswaterman, 2024-05-08:
It seems you've gotten to the bottom of it.  Thank you for doing nearly all of the detective work yourself!