riscv-software-src/riscv-isa-sim #1665: mstatus.fs cannot be written as initial

omerguzelelectronicguy, 2024-05-08:
I am trying the different coverage cases for a RISC-V CPU. I have done the simulation below in Spike.

```
core   0: >>>>  change_ms
core   0: 0x00000000800000d8 (0x00d19213) slli    tp, gp, 13
core   0: 3 0x00000000800000d8 (0x00d19213) x4  0x0000000000002000
(spike) 
core   0: 0x00000000800000dc (0x000001ba) c.slli  gp, 14
core   0: 3 0x00000000800000dc (0x01ba) x3  0x0000000000004000
(spike) 
core   0: 0x00000000800000de (0x00018863) beqz    gp, pc + 16
core   0: 3 0x00000000800000de (0x00018863)
(spike) 
core   0: 0x00000000800000e2 (0x3001a273) csrrs   tp, mstatus, gp
core   0: 3 0x00000000800000e2 (0x3001a273) x4  0x8000000a00006000 c768_mstatus 0x8000000a00006000
(spike) 
core   0: 0x00000000800000e6 (0x30023273) csrrc   tp, mstatus, tp
core   0: 3 0x00000000800000e6 (0x30023273) x4  0x8000000a00006000 c768_mstatus 0x0000000a00000000
```

At the simulation, I was trying to change the fs from `dirty` to clean. However, it changes from `dirty` to `Off`. Is there any point I missed from the spec that causes this behaviour or is it a bug in spike?

aswaterman, 2024-05-08:
It looks like x4 has both bits 14..13 set at the time you’re clearing the CSR, so you’re actually attempting to change FS from Dirty to Off, not Initial, so Spike is doing the right thing.

omerguzelelectronicguy, 2024-05-13:
Thank you. I missed it. I'm sorry for causing you unnecessary reporting and taking your time.

closed 2024-05-13 by omerguzelelectronicguy