riscv-software-src/riscv-isa-sim #1668: Why misaligned store will raise store/amo access fault at the function store_slow_path?

asker-qiao, 2024-05-15:
I view the source file at riscv/mmu.cc:300, the store or amo instruction at here will trap store/amo access fault rather misaligned fault, but the condition of if is require misaligned. 

the source code:
```c++
// mmu.cc
299:  if (require_alignment)
300:        throw trap_store_access_fault(gva, addr, 0, 0);
```

yangye0212, 2024-05-21:
Ok, thank you, I see that ISA spec V20211203 at 3.6.3.3 Alignment, the section specified implementation may raise access fault when amo or lr/sc instruction address is misaligned. So I think this implementation is a optional way to do, is right? And the signification of this way is tell trap handler should not be emulated.

aswaterman, 2024-05-21:
Yeah.

closed 2024-05-21 by yangye0212