riscv-software-src/riscv-isa-sim #1669: Zfa extension instruction fcvtmod_w_d behavior conflict with sail model on too large/ too small boundary

GuoShibo-cn, 2024-05-16:
```
  int true_exp = exp - 1023;
  int shift = true_exp - 52;

  /* Restore implicit bit.  */
  frac |= 1ull << 52;

  /* Shift the fraction into place.  */
  if (shift >= 64) {
    /* The fraction is shifted out entirely.  */
    frac = 0;
  } else  if ((shift >= 0) && (shift < 64)) {
    /* The number is so large we must shift the fraction left.  */
    frac <<= shift;
  } else if ((shift > -64) && (shift < 0)) {
    /* Normal case -- shift right and notice if bits shift out.  */
    inexact = (frac << (64 + shift)) != 0;
    frac >>= -shift;
  } else {
    /* The fraction is shifted out entirely.  */
    frac = 0;
    inexact = true;
  }
```
According to spike fcvtmod_w_d.h too large/ too small boundary true_exp is shift+52 which is (64 + 52 = 116)/ (-64 + 52 = -12) however according to sail model [sail model](https://github.com/riscv/sail-riscv/blob/master/model/riscv_insts_zfa.sail#L762)
```
  let is_too_large  = true_exp >= 84;  /* bits will be 'multiplied' out */
  let is_too_small  = true_exp <  0;   /* bits will be 'divided' out    */

       if is_zero       then (zeros(),  zeros())
  else if is_subnorm    then (nxFlag(), zeros())
  else if is_nan_or_inf then (nvFlag(), zeros())
  else if is_too_large  then (nvFlag(), zeros())
  else if is_too_small  then (nxFlag(), zeros())
  else {
```
boundary true_exp is 84/0 
which model could be considered as meet specification?

aswaterman, 2024-05-16:
Can you provide a particular input where Spike and the SAIL model disagree?

nibrunieAtSi5, 2024-05-16:
I am not a SAIL expert but it seems the implementation differs in how they handle the building of a temporary value (84 bits fixed point for SAIL while spike uses a 64-bit supporting integer type for the integer part).