riscv-software-src/riscv-isa-sim #1672: Build fails with "error: 'L_tmpnam' was not declared in this scope"

k138z, 2024-05-17:
I am on Ubuntu 22.04.4 LTS. I tried building Spike on both commit hashes 20a2b6d0 (what riscv-gnu-toolchain has) and the latest master (c81d8e73).
My configure arguments are as follows:
`../configure --prefix=$RISCV --host=riscv64-unknown-linux-gnu`

Here is what I get on `make`:

```
riscv64-unknown-linux-gnu-g++ -MMD -MP  -DPREFIX=\"\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -std=c++2a -g -O2   -iquote . -I.. -iquote ../fesvr -iquote ../riscv -iquote ../disasm -iquote ../customext -iquote ../fdt -iquote ../softfloat -iquote ../spike_main -iquote ../spike_dasm -I/usr/include   -c ../fesvr/elfloader.cc
In file included from /opt/riscv/riscv64-unknown-linux-gnu/include/c++/13.2.0/cstdio:42,
                 from /opt/riscv/riscv64-unknown-linux-gnu/include/c++/13.2.0/ext/string_conversions.h:45,
                 from /opt/riscv/riscv64-unknown-linux-gnu/include/c++/13.2.0/bits/basic_string.h:4097,
                 from /opt/riscv/riscv64-unknown-linux-gnu/include/c++/13.2.0/string:54,
                 from /opt/riscv/riscv64-unknown-linux-gnu/include/c++/13.2.0/stdexcept:39,
                 from ../fesvr/memif.h:8,
                 from ../fesvr/elfloader.cc:5:
/usr/include/stdio.h:205:27: error: 'L_tmpnam' was not declared in this scope
  205 | extern char *tmpnam (char[L_tmpnam]) __THROW __wur;
      |                           ^~~~~~~~
/usr/include/stdio.h:210:33: error: 'L_tmpnam' was not declared in this scope; did you mean 'tmpnam'?
  210 | extern char *tmpnam_r (char __s[L_tmpnam]) __THROW __wur;
      |                                 ^~~~~~~~
      |                                 tmpnam
cc1plus: note: unrecognized command-line option '-Wno-nonportable-include-path' may have been intended to silence earlier diagnostics
```

All help appreciated, thanks!

jerryz123, 2024-05-17:
Drop the --host argument. I assume you are trying to compile Spike as a native x86 for your workstation.

k138z, 2024-05-20:
Sorry, yes. I had a fundamental misunderstanding about how all of this works - is not an issue now. Closing.

closed 2024-05-20 by k138z

tianboh, 2024-05-20:
Hi there, I met a similar problem, though I do not have host option. I am trying to compile Spike as a native binary as @jerryz123 said `../configure --prefix=$RISCV`

The error is
```
g++ -MMD -MP  -DPREFIX=\"/opt/riscv\" -Wall -Wno-unused -Wno-nonportable-include-path -g -O2 -fPIC -std=c++2a -g -O2   -iquote . -I.. -iquote ../fesvr -iquote ../riscv -iquote ../disasm -iquote ../customext -iquote ../fdt -iquote ../softfloat -iquote ../spike_main -iquote ../spike_dasm -I/include/boost-0  -fPIC -I../fdt -c ../riscv/csrs.cc
In file included from ../riscv/csrs.h:8,
                 from ../riscv/csrs.cc:6:
../riscv/csrs.cc: In member function ‘virtual reg_t dcsr_csr_t::read() const’:
../riscv/csrs.cc:1315:30: error: ‘DCSR_STOPCYCLE’ was not declared in this scope; did you mean ‘DCSR_STOPCOUNT’?
 1315 |   result = set_field(result, DCSR_STOPCYCLE, 0);
      |                              ^~~~~~~~~~~~~~
../riscv/decode.h:235:52: note: in definition of macro ‘set_field’
  235 |   (((reg) & ~(std::remove_cv<decltype(reg)>::type)(mask)) | (((std::remove_cv<decltype(reg)>::type)(val) * ((mask) & ~((mask) << 1))) & (std::remove_cv<decltype(reg)>::type)(mask)))
      |                                                    ^~~~
../riscv/csrs.cc: In member function ‘virtual bool dcsr_csr_t::unlogged_write(reg_t)’:
../riscv/csrs.cc:1334:25: error: ‘DCSR_HALT’ was not declared in this scope; did you mean ‘DCSR_PELP’?
 1334 |   halt = get_field(val, DCSR_HALT);
      |                         ^~~~~~~~~
../riscv/decode.h:232:51: note: in definition of macro ‘get_field’
  232 |   (((reg) & (std::remove_cv<decltype(reg)>::type)(mask)) / ((mask) & ~((mask) << 1)))
      |                                                   ^~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-nonportable-include-path’
make: *** [Makefile:347: csrs.o] Error 1
```
My $RISCV has following files, which is generated from `ct-ng riscv64-unknown-elf`
```
bin  build.log.bz2  include  lib  libexec  riscv64-unknown-elf  share
```

tianboh, 2024-05-20:
The problem is encoding.h is not updated in my local environment. Once updated, problem fixed.