riscv-software-src/riscv-isa-sim #1676: Facing Issue running Zvbb RVV Cryptography Instructions

akifejaz, 2024-05-23:
I'm trying to run the Vector Bit-manipulation Zvbb on spike but I'm facing issue. Spike says 'trap_illegal_instruction', this is how I'm running test, I'm I missing something?
`spike --isa=RV64gcV -d --varch=vlen:2048,elen:64 -m0x100000000:0x800000000 unary_crypto_m1_e32_md1_0/test.elf` 

[image]

Can you please help me with this please?
@arunthomas @scottj97

aswaterman, 2024-05-27:
The ISA string you specified doesn't include Zvbb.  Try using `--isa=rv64gcv_zvbb` instead.

akifejaz, 2024-05-27:
> The ISA string you specified doesn't include Zvbb. Try using `--isa=rv64gcv_zvbb` instead.

Is there a single ISA flag for enabling whole crypto since there are different extensions under crypto spec. as in image below. For example. VCLZ works when I'm adding `--isa=rv64gcv_zvbb` and VAESEF works when I'm adding `--isa=rv64gcv_zvkend` Can I add single ISA flag for all these instructions supported? Thanks alot.

[image]

aswaterman, 2024-05-27:
You’ll have to consult the ISA specs to see which extensions include which instructions. Spike conforms to that spec.

closed 2024-05-27 by aswaterman