riscv-software-src/riscv-isa-sim #1681: Vector Widening Integer Multiply-Add Instructions - not working

deepaannnavi, 2024-05-28:
> worked in vector widening mac instruction 

[image]
set element width as e = 64, LMUL =1
loading the 32 bit element to v1 vector register
loading the 32 bit element to v0 vector register
loading the 64 bit element to v8 vector register
vwmaccu.vv do the unsigned multiplication(v0*v8) and add the result to v8 vector register and store the result to the v8 vector register

> disassembly file output:

[image]

> Dump file output shows as:

[image]
 
here v0,v1,v8 vector register got loaded, but the vector widening Integer multiply add instruction doesn't work.
It throws mcause 2 which tells it is illegal instruction. 
can anyone resolve this and help me telling how to make this instruction work without error

**Note**

> spike commit :[a53a71f](https://github.com/riscv-software-src/riscv-isa-sim/commit/a53a71fcc3c985cf95973e86e40814c30c551a68)

>command: spike --log-commits --log spike.dump +signature=spike_signature.txt ${Test_file_NAME}.elf`

aswaterman, 2024-05-28:
Widening multiplies use SEW for the inputs and 2xSEW for the outputs. So you're trying to do a 64->128 multiply, which is not supported.  You need to set SEW=32.

deepaannnavi, 2024-05-29:
Thank you so much it worked now

> risc v assembly code:
[image]

> disassembly file
[image]

> dump file:

[image]

closed 2024-05-29 by deepaannnavi