riscv-software-src/riscv-isa-sim #1685: assert failed in vectorUnit.cc from 71line. T&vectorUnit_t::elt(reg_t,reg_t,bool)[with T=long unsigned int; reg_t=long unsigned int]: Assertion '(VLEN>>3)/sizeof(T)>0' failed.

huihuiei, 2024-06-03:
in my testbench, spike is as reference model through dpi interface into my entirement. I set "vlen:32,elen:32", in static cfg_t cfg(std::make_pair(0,0)......) function. but the simulation will stop because the error: T&vectorUnit_t::elt(reg_t,reg_t,bool)[with T=long unsigned int; reg_t=long unsigned int]: Assertion '(VLEN>>3)/sizeof(T)>0' failed.
except the cfg_t cfg function, which function should i change or transfer?

closed 2024-06-03 by aswaterman

huihuiei, 2024-06-05:
i think you could replicate this problem. set varch 32,32. then execute: vsetivli a2, 0x1, e32,  vmv.v.x v0, t1; then there will be the same error.

aswaterman, 2024-06-05:
We haven’t put effort into supporting VLEN=ELEN configs, so it’s at least conceivable this is a limitation of vanilla Spike. But I don’t have resources to look into it for now. Submit a PR if you can.

cc @chihminchao