riscv-software-src/riscv-isa-sim #1692: Dynamic CSR read/write mask

ved-rivos, 2024-06-13:
Some CSR fields  may be read-only zero if the extension providing them is disabled in the controlling `*envcfg`  CSR. Presently CSR read/write masks seem to be built statically based on whether the extension is enabled. Looking for some suggestion on best way to add the dynamic masking.

aswaterman, 2024-06-20:
It is true that `masked_csr_t` requires the mask be static, but you can instead inherit directly from `csr_t` and implement your own custom read-masking function.  This way, the masking can be a function of other dynamic state.  For an example, see `epc_csr_t`, which forms its read mask based upon IALIGN, which can change at runtime when the C extension is disabled.

ved-rivos, 2024-06-20:
Thanks. I will follow that form for mstatus and sstatus.

closed 2024-06-20 by ved-rivos