riscv-software-src/riscv-isa-sim #1696: Legal ISA string option not accepted

christian-herber-nxp, 2024-06-19:
These strings are accepted:

```
spike  --isa=rv32im_zce_zilsd_zcmlsd
spike  --isa=rv32im_za_zilsd_zcmlsd
```

While this isn't:

```
spike  --isa=rv32imc_zilsd_zcmlsd_zfinx
error: bad --isa option 'rv32imc_zilsd_zcmlsd_zfinx'. 'Zcmlsd' extension requires 'Zca' and 'Zilsd' extensions
```

From the isa_parser, I would expect that the second one should also work, as having C should assert Zca:

```
  if (extension_table['C']) {
    extension_table[EXT_ZCA] = true;
    if (extension_table['F'] && max_xlen == 32)
      extension_table[EXT_ZCF] = true;
    if (extension_table['D'])
      extension_table[EXT_ZCD] = true;
  }
```

It seems that extension_table['C'] is not set anywhere in the isa_parser.

aswaterman, 2024-06-20:
This does appear to be a bug.  Would you like to take a stab at this one, or should one of us handle it later this week?

christian-herber-nxp, 2024-06-20:
I couldn't find the code that is supposed to set extension_table['C'] == true, so honestly, I have no idea where to look to fix this.

aswaterman, 2024-06-20:
OK, I'll take this one.

aswaterman, 2024-06-20:
For future reference, this is where `extension_table['C']` is set: https://github.com/riscv-software-src/riscv-isa-sim/blob/5e7928a3475f3fe6a148abd74abb6d97c65c31f2/disasm/isa_parser.cc#L78

So the problem is just that the logic that handles C -> Zca occurs too late in the procedure; it needs to be hoisted up before the error-checking code.

PR #1702 (2024-06-21): In isa_parser, move extensionology code before error-checking code
Resolves #1696
commit c6cb05ce (2024-06-21): In isa_parser, move extensionology code before error-checking code

Resolves #1696
closed 2024-06-21 by aswaterman