riscv-software-src/riscv-isa-sim #1727: Deprecated dcsr.halt

YenHaoChen, 2024-07-15:
Hello,

To my knowledge, the dcsr_csr_t::halt variable is for halt_on_reset, i.e., the -H flag for the debugger. Spike leaves a backdoor for altering the halt variable, i.e., writing a read-only bit, dcsr.nmip (riscv/csrs.cc:1357). 
```  halt = get_field(val, DCSR_NMIP);```

This backdoor isn't generally specified in the specification. Is this backdoor typically required for the openOCD?
@rtwfroody Would you share some words about this?

(A branch decouples the halt variable from the dcsr.nmip. https://github.com/riscv-software-src/riscv-isa-sim/pull/1726)

Best regards,
YenHaoChen

rtwfroody, 2024-07-16:
dcsr.halt is ancient, and nowhere to be found in any recent debug spec. (E.g. I have a 2019 version here that doesn't include include this bit.) AFAIK OpenOCD does not require this, and if it does then it's a bug in OpenOCD.

YenHaoChen, 2024-07-17:
@rtwfroody Thanks for your clarification.

closed 2024-07-17 by YenHaoChen