riscv-software-src/riscv-isa-sim #1745: Why  menvcfg.pbmt is set 1 on reset ?

chihminchao, 2024-07-24:
It is the only bit initialized as 1 if extension is implemented.  However, I can't find the related statement in spec
https://github.com/riscv-software-src/riscv-isa-sim/blob/64bc0c1f1d8ff3746ef4eb39e9d142a4456ed029/riscv/processor.cc#L394

The change is introduced by https://github.com/riscv-software-src/riscv-isa-sim/pull/992/commits/ea70a9359daf5780b815b68aa9cdc7b2a71d2f8c

@rbuchner-aril may you comment ?

aswaterman, 2024-07-24:
Based on the commit message, it looks like @rbuchner-aril did this because Spike previously enabled PBMTE unconditionally, and so making the reset value 1 prevented backwards incompatibility with respect to older Spike.

I agree this is surprising behavior (though not incorrect, since the reset value is unspecified). I’d be OK with a PR that resets this field to 0 instead.

rbuchner-aril, 2024-07-24:
Yes, what Andrew said is correct.

aswaterman, 2024-07-25:
Thanks for the confirmation, @rbuchner-aril.  @chihminchao if you prepare a PR to make the change, I'll merge it.

chihminchao, 2024-07-25:
The PR has been sent
https://github.com/riscv-software-src/riscv-isa-sim/pull/1746

aswaterman, 2024-07-25:
Resolved by #1746.

closed 2024-07-25 by aswaterman