riscv-software-src/riscv-isa-sim #1747: Need assistance for Executing my targeted workload on spike RISC-V ISA after adding custom trigonometric Instructions

baul-iisc, 2024-07-25:
Hello team,
I have added the trigonometric instructions on riscv-gnu-toolchain. Now I am specifically looking for the steps required/process to compile my workload in Spike and Gem5 with these newly added instructions and then characterize the workload to evaluate the performance impact of these instructions.
I would greatly appreciate it if someone could provide guidance on how to compile code that utilizes these newly added instructions and execute it on Spike pk to characterize the workload.
Thank you in advance for your assistance.

TommyMurphyTM1234, 2024-07-25:
> I would greatly appreciate it if someone could provide guidance on how to compile code that utilizes these newly added instructions and execute it on Spike pk to characterize the workload. 

As I said here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1518#issuecomment-2249763994

you can use inline assembly to compile your custom instructions into a RISC-V executable using a RISC-V toolchain that has been extended to support those instructions.

You'll probably need to explain in more detail what exactly you mean by this: 

> execute it on Spike pk to characterize the workload. 

What sort of "characterization" do you mean?

It's also confusing that you mention Spike and Gem5 when your focus seems to be specifically on Spike (I think)?

ved-rivos, 2024-07-25:
Here are some examples of unit tests that can be run on spike: https://github.com/riscv-software-src/riscv-tests

baul-iisc, 2024-07-25:
I want to characterize some special scientific computing workload which is mix of floating point computation and lots of trigonometric operation on it. These trigonometric operations are right now implemented using software library. Now with these trigonometric hardware implementation if this computation getting benefits of execution time? if so how much % it is getting improvement, then based on that I can take decision if I can really go for trigonometric hardware implementation or not. Actually these algorithms are huge line of code, so it will not be an easy job to modify the code using inline assembly.

TommyMurphyTM1234, 2024-07-25:
> Given the substantial size and complexity of these algorithms, modifying the code to use inline assembly would be a significant undertaking.

In that case then you'll probably have to modify the compiler so that it can generate your custom instructions when it detects appropriate intermediate code patterns. This is not a trivial exercise for someone not familiar with compiler toolchain internals. You'll probably need to ask in an appropriate GCC and/or LLVM/Clang (depending on what toolchain you're going to use) forum. There are mailing lists and other discussion forums for these projects that may be relevant to this topic.

baul-iisc, 2024-07-27:
Hello team,
I am trying to add custom trigonometric instructions on spike riscv-gnu-toolchain to execute my workload and do an assesment of performance improvement with this newly added instructions, but when I gave the command sudo make build-sim, it it throwing errors. PLease suggest if there is any mistake in the following process/ or if I am missing something. THse trigonometric instructions I want use similar to the single and double precision fsqrt instruction.  The steps I am following is given below, please let me know where i am making any mistake or missing anything:

1. I added the MASK_ins and MATCH_ins to riscv-gnu-toolchain/riscv-binutils/include/opcode/riscv-opc.h.
#define MATCH_GCD 0x6027
#define MASK_GCD 0xfe00707f
#define MATCH_FACT 0x2b
#define MASK_FACT 0x7f
#define MATCH_MOD 0x200006b
#define MASK_MOD 0xfe00707f
#define MATCH_FSIN_S 0x59000053
#define MASK_FSIN_S  0xfff0007f
#define MATCH_FCOS_S 0x5b000053
#define MASK_FCOS_S  0xfff0007f
#define MATCH_FTAN_S 0x5d000053
#define MASK_FTAN_S  0xfff0007f
#define MATCH_FSIN_D 0x53000053
#define MASK_FSIN_D  0xfff0007f
#define MATCH_FCOS_D 0x55000053
#define MASK_FCOS_D  0xfff0007f
#define MATCH_FTAN_D 0x57000053
#define MASK_FTAN_D  0xfff0007f

#ifdef DECLARE_INSN
DECLARE_INSN(mod, MATCH_MOD, MASK_MOD)
DECLARE_INSN(gcd, MATCH_GCD, MASK_GCD)
DECLARE_INSN(fact, MATCH_FACT, MASK_FACT)
DECLARE_INSN(fsin_s, MATCH_FSIN_S, MASK_FSIN_S)
DECLARE_INSN(fcos_s, MATCH_FCOS_S, MASK_FCOS_S)
DECLARE_INSN(ftan_s, MATCH_FTAN_S, MASK_FTAN_S)
DECLARE_INSN(fsin_d, MATCH_FSIN_D, MASK_FSIN_D)
DECLARE_INSN(fcos_d, MATCH_FCOS_D, MASK_FCOS_D)
DECLARE_INSN(ftan_d, MATCH_FTAN_D, MASK_FTAN_D)

2. Then, in riscv-gnu-toolchain/riscv-binutils/opcodes/riscv-opc.c I add the following content:

{"mod",      0, INSN_CLASS_I,       "d,s,t",     MATCH_MOD, MASK_MOD, match_opcode, 0 },
{"gcd",      0, INSN_CLASS_I,       "d,s,t",     MATCH_GCD, MASK_GCD, match_opcode, 0 },
{"fact",     0, INSN_CLASS_I,       "d,a",       MATCH_FACT, MASK_FACT, match_opcode, 0 },
// Single-precision floating-point instruction subset
{"fsin.s",    0, INSN_CLASS_F_INX,   "D,S",       MATCH_FSIN_S|MASK_RM, MASK_FSIN_S|MASK_RM, match_opcode, 0 },
{"fsin.s",    0, INSN_CLASS_F_INX,   "D,S,m",     MATCH_FSIN_S, MASK_FSIN_S, match_opcode, 0 },
{"fcos.s",    0, INSN_CLASS_F_INX,   "D,S",       MATCH_FCOS_S|MASK_RM, MASK_FCOS_S|MASK_RM, match_opcode, 0 },
{"fcos.s",    0, INSN_CLASS_F_INX,   "D,S,m",     MATCH_FCOS_S, MASK_FCOS_S, match_opcode, 0 },
{"ftan.s",    0, INSN_CLASS_F_INX,   "D,S",       MATCH_FTAN_S|MASK_RM, MASK_FTAN_S|MASK_RM, match_opcode, 0 },
{"ftan.s",    0, INSN_CLASS_F_INX,   "D,S,m",     MATCH_FTAN_S, MASK_FTAN_S, match_opcode, 0 },

/* Double-precision floating-point instruction subset.  */
{"fsin.d",    0, INSN_CLASS_D_INX,   "D,S",       MATCH_FSIN_D|MASK_RM, MASK_FSIN_D|MASK_RM, match_opcode, 0 },
{"fsin.d",    0, INSN_CLASS_D_INX,   "D,S,m",     MATCH_FSIN_D, MASK_FSIN_D, match_opcode, 0 },
{"fcos.d",    0, INSN_CLASS_D_INX,   "D,S",       MATCH_FCOS_D|MASK_RM, MASK_FCOS_D|MASK_RM, match_opcode, 0 },
{"fcos.d",    0, INSN_CLASS_D_INX,   "D,S,m",     MATCH_FCOS_D, MASK_FCOS_D, match_opcode, 0 },
{"ftan.d",    0, INSN_CLASS_D_INX,   "D,S",       MATCH_FTAN_D|MASK_RM, MASK_FTAN_D|MASK_RM, match_opcode, 0 },
{"ftan.d",    0, INSN_CLASS_D_INX,   "D,S,m",     MATCH_FTAN_D, MASK_FTAN_D, match_opcode, 0 },

3. After that , I recompiled my riscv-gnu-toolchain.
4. To achieve command functionality, I want to simulate these trigonometric instructions in Spike and gcc.So I firstly add MATCH_ins and MASK_ins in spike/riscv/encoding.h,
#define MATCH_GCD 0x6027
#define MASK_GCD 0xfe00707f
#define MATCH_FACT 0x2b
#define MASK_FACT 0x7f
#define MATCH_MOD 0x200006b
#define MASK_MOD 0xfe00707f
#define MATCH_FSIN_S 0x59000053
#define MASK_FSIN_S  0xfff0007f
#define MATCH_FCOS_S 0x5b000053
#define MASK_FCOS_S  0xfff0007f
#define MATCH_FTAN_S 0x5d000053
#define MASK_FTAN_S  0xfff0007f
#define MATCH_FSIN_D 0x53000053
[... 158 lines cut ...]
 1167 |   #define DECLARE_OVERLAP_INSN(name, ext) { name##_supported = isa->extension_enabled(ext); }
      |                                             ^~~~
/Data/chandra/development/RISCV-SPIKE/RISCV-BUILD/riscv-gnu-toolchain/spike/riscv/overlap_list.h:21:22: error: ‘rstsa32_supported’ was not declared in this scope; did you mean ‘xperm32_supported’?
   21 | DECLARE_OVERLAP_INSN(rstsa32, EXT_ZPN)
      |                      ^~~~~~~
/Data/chandra/development/RISCV-SPIKE/RISCV-BUILD/riscv-gnu-toolchain/spike/riscv/processor.cc:1167:45: note: in definition of macro ‘DECLARE_OVERLAP_INSN’
 1167 |   #define DECLARE_OVERLAP_INSN(name, ext) { name##_supported = isa->extension_enabled(ext); }
      |                                             ^~~~
/Data/chandra/development/RISCV-SPIKE/RISCV-BUILD/riscv-gnu-toolchain/spike/riscv/overlap_list.h:22:22: error: ‘srli32_u_supported’ was not declared in this scope; did you mean ‘vle32_v_supported’?
   22 | DECLARE_OVERLAP_INSN(srli32_u, EXT_ZPN)
      |                      ^~~~~~~~
/Data/chandra/development/RISCV-SPIKE/RISCV-BUILD/riscv-gnu-toolchain/spike/riscv/processor.cc:1167:45: note: in definition of macro ‘DECLARE_OVERLAP_INSN’
 1167 |   #define DECLARE_OVERLAP_INSN(name, ext) { name##_supported = isa->extension_enabled(ext); }
      |                                             ^~~~
[... cut ...]
/Data/chandra/development/RISCV-SPIKE/RISCV-BUILD/riscv-gnu-toolchain/spike/riscv/processor.cc:1167:45: note: in definition of macro ‘DECLARE_OVERLAP_INSN’
 1167 |   #define DECLARE_OVERLAP_INSN(name, ext) { name##_supported = isa->extension_enabled(ext); }
      |                                             ^~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-nonportable-include-path’
make[1]: *** [Makefile:347: processor.o] Error 1
make[1]: Leaving directory '/Data/chandra/development/RISCV-SPIKE/RISCV-BUILD/riscv-gnu-toolchain/build-spike'
make: *** [Makefile:901: stamps/build-spike] Error 2

closed 2025-02-14 by baul-iisc