riscv-software-src/riscv-isa-sim #1752: Segmentation Fault when interrupt handled while interrupts are disabled in machine mode

mkorbel1, 2024-07-31:
When running in machine mode `PRV_M`, and an interrupt is handled by `take_interrupt` while interrupts are disabled, the current implementation attempts to access `sstatus`, which only exists in supervisor mode.  This results in a segmentation fault.

An example use case for this would be running in machine mode, interrupts disabled, then a WFI which waits for a configured timer to trigger an interrupt by setting `mtimecmp`.

Code snippet from processor.cc:
```C++
void processor_t::take_interrupt(reg_t pending_interrupts)
{
  // Do nothing if no pending interrupts
  if (!pending_interrupts) {
    return;
  }

  // Exit WFI if there are any pending interrupts
  in_wfi = false;

  // M-ints have higher priority over HS-ints and VS-ints
  const reg_t mie = get_field(state.mstatus->read(), MSTATUS_MIE);
  const reg_t m_enabled = state.prv < PRV_M || (state.prv == PRV_M && mie);
  reg_t enabled_interrupts = pending_interrupts & ~state.mideleg->read() & -m_enabled;
  if (enabled_interrupts == 0) {
    // HS-ints have higher priority over VS-ints
    const reg_t deleg_to_hs = state.mideleg->read() & ~state.hideleg->read();
    const reg_t sie = get_field(state.sstatus->read(), MSTATUS_SIE);
    const reg_t hs_enabled = state.v || state.prv < PRV_S || (state.prv == PRV_S && sie);
    enabled_interrupts = pending_interrupts & deleg_to_hs & -hs_enabled;
    if (state.v && enabled_interrupts == 0) {
      // VS-ints have least priority and can only be taken with virt enabled
      const reg_t deleg_to_vs = state.hideleg->read();
      const reg_t vs_enabled = state.prv < PRV_S || (state.prv == PRV_S && sie);
      enabled_interrupts = pending_interrupts & deleg_to_vs & -vs_enabled;
    }
  }
```

A simple fix for this issue might be to just qualify this `if` with `PRV_S` like this:

```C++
   if (enabled_interrupts == 0 && state.prv == PRV_S) {
```

In the described example test case, this resolved the issue.

Previously cloned versions of spike (approximately late year) did not hit this issue, but I could not find what broke the functionality from some looking through git history.

aswaterman, 2024-07-31:
Thanks for the bug report.  It's not that sstatus only exists in S-mode; it's that it exists if S-mode is implemented.  I'll make a PR that fixes it in that manner.

closed 2024-08-01 by aswaterman

mkorbel1, 2024-08-01:
Thank you @aswaterman for the very fast turnaround on the fix!

aswaterman, 2024-08-01:
You're welcome!

commit 584f8551 (2024-07-31): Fix segfault accessing menvcfg when U-mode doesn't exist

The simplest fix is to create the CSRs even if they don't need to exist,
and just skip adding them to the CSR map to prevent the target machine
from being able to access them.

It looks like there are other place we should be following this pattern:
e.g. why does sstatus exist if S-mode does not exist?  But that's a
matter for another day.

Resolves #1752
PR #1753 (2024-08-01): Fix segfault accessing menvcfg when U-mode doesn't exist
The simplest fix is to create the CSRs even if they don't need to exist, and just skip adding them to the CSR map to prevent the target machine from being able to access them.

It looks like there are other place we should be following this pattern: e.g. why does sstatus exist if S-mode does not exist?  But that's a matter for another day.

Resolves #1752