riscv-software-src/riscv-isa-sim #1762: Spike does not implement the functions of VTA and MTA.

Popebl, 2024-08-06:
According to the RISC-V 'V' Vector Extension Version 1.0, section 3.4.3, Vector Tail Agnostic (VTA) and Vector Mask Agnostic (VMA) are defined. The RISC-V specification outlines the functions of VTA and VMA.

In the following code, Spike extracts the VTA and VMA values from the instruction.

```c
reg_t vectorUnit_t::vectorUnit_t::set_vl(int rd, int rs1, reg_t reqVL, reg_t newType)
{
  int new_vlmul = 0;
  if (vtype->read() != newType) {
    vsew = 1 << (extract64(newType, 3, 3) + 3);
    new_vlmul = int8_t(extract64(newType, 0, 3) << 5) >> 5;
    vflmul = new_vlmul >= 0 ? 1 << new_vlmul : 1.0 / (1 << -new_vlmul);
    vlmax = (VLEN/vsew) * vflmul;
    vta = extract64(newType, 6, 1);
    vma = extract64(newType, 7, 1);

    vill = !(vflmul >= 0.125 && vflmul <= 8)
           || vsew > std::min(vflmul, 1.0f) * ELEN
           || (newType >> 8) != 0;

    if (vill) {
      vlmax = 0;
      vtype->write_raw(UINT64_MAX << (p->get_xlen() - 1));
    } else {
      vtype->write_raw(newType);
    }
  }
```
However, Spike does not use either in the function code. Please add the function code for Spike.

Thanks very much!

aswaterman, 2024-08-06:
Spike correctly implements the ISA.  The ISA says that, when vta=1, tail elements are either undisturbed or are written with 1s.  (Likewise vma and masked elements.)  Therefore, ignoring the vta and vma fields is a correct implementation.

Popebl, 2024-08-07:
Thanks very much

Popebl, 2024-08-07:
Thanks very much

aswaterman, 2024-08-07:
You're welcome!

Popebl, 2024-08-07:
Will Spike support the implementation of "Written with 1s"?

aswaterman, 2024-08-07:
See past discussion at https://github.com/riscv-software-src/riscv-isa-sim/pull/1593

The main problem is realistic microarchitectures would almost never use the "write 1s" strategy all the way to the end of the vector register group.  The more typical strategy is to write 1s up until a particular microarchitecture-specific boundary (say, the next VLEN-bit boundary, or maybe VLEN/2 or VLEN*2), then use the undisturbed policy for the remainder of the tail.  So it's not just a Boolean property: we'd need a richer way of describing the behavior in order to make the feature useful.

Popebl, 2024-08-07:
Get it. Thanks very much!

closed 2024-08-07 by Popebl