riscv-software-src/riscv-isa-sim #1767: How to change vector extension configuration(vlen, elen)?

Seo-sang, 2024-08-09:
Hello, I have a problem when building spike with --with-varch option and I found varch option is removed two month ago. Then, how to change vector configuration? Thank you.

[image]

jerryz123, 2024-08-09:
This is done at runtime with the zvl/zve isa strings now. 

For example, you can specify --isa=rv64gcv_zvl512b_zve64d 

No flags need to be passed to the configure script.

Seo-sang, 2024-08-09:
@jerryz123 I have one more question. Even though I set --isa=rv64gcv_zvl128b_zve32d, elen doesn't changed and it seems that elen cannot be smaller than 64-bit in isa_parser.cc file line 349. How can I set elen to 32-bit?
Thank you.

jerryz123, 2024-08-11:
`v` extension requires `zve64d`, and `zve32f` is a subset of `zve64d` (`zve32f` does not imply `zve64d` is not available).

You can do `rv64gc_zvl128b_zve32f` to get a machine with ELEN=32.

nadime15, 2025-03-17:
Hi,

I was curious about how to configure VLEN and ELEN and came across this post. I looked into the code for setting ELEN:

https://github.com/riscv-software-src/riscv-isa-sim/blob/3f79e3b7ded80d9ef0e722126b3765207e010711/disasm/isa_parser.cc#L359-L377

It looks like ELEN can only be set to 32 or 64 (with 64 as the default), meaning it's not possible to configure it to, say, 128. The code explicitly checks that ELEN is either 32 or 64.

As far as I understand, EEW doesn’t need to be equal to ELEN, but it sets a lower bound for ELEN. However, doesn’t the current implementation effectively constrain ELEN? I can have EEW = 64, ELEN = 128 and VLEN = 256

However, for VLEN, there isn’t a strict check for specific values, just a final validation ensuring VLEN ≤ 4096b. Looking at how Zvl is handled, it seems like I could theoretically specify Zvl2048b (even though this extension does not exist yet), and it would pass, right?