riscv-software-src/riscv-isa-sim #1773: `fmv.w.x` is illegal?

Donkey-Hao, 2024-08-14:
Hi, I am learning the risc-v ISA and spike. Last day I saw the issue #1765 and tried to run it:
```asm:
.section .data
.section .text
.globl _start

_start:
    fmv.w.x ft4, zero
    fmv.w.x ft5, zero
    fsqrt.s ft5, ft4
```
I assembled it and linked it, then run spike I got this:
```
(base) ➜ spike -d prog                
warning: tohost and fromhost symbols not in ELF; can't communicate with target
(spike) r 5
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   0: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
core   0: 0x0000000000001010 (0x00028067) jr      t0
(spike) r 1
core   0: >>>>  _start
core   0: 0x0000000080000000 (0xf0000253) fmv.w.x ft4, zero
core   0: exception trap_illegal_instruction, epc 0x0000000080000000
core   0:           tval 0x00000000f0000253
(spike) q
```
I checked the ISA manual and compared it with that issue (1765). I'm very confused, why the code shows error. The spike version is:
```
(base) ➜ spike --help
Spike RISC-V ISA Simulator 1.1.1-dev
```

DavidYu360, 2024-08-14:
1. Does the default setting support FPU instructions?
2. If 1 is true, check the value of CSR field `mstatus.FS` to see if it is off.

aswaterman, 2024-08-14:
This is definitely not a Spike bug. As @DavidYu360 suggests, this could be because FS=0.

closed 2024-08-14 by aswaterman

Donkey-Hao, 2024-08-14:
> 1. Does the default setting support FPU instructions?
> 
> 2. If 1 is true, check the value of CSR field `mstatus.FS` to see if it is off.

So that's how it is. Learned. Thank you all.

Donkey-Hao, 2024-08-28:
Hi @DavidYu360, @aswaterman, 
Sorry to bother you again.
The default ISA is `  --isa=<name>          RISC-V ISA string [default RV64IMAFDC_zicntr_zihpm]`.  
> 1. Does the default setting support FPU instructions?
> 
> 2. If 1 is true, check the value of CSR field `mstatus.FS` to see if it is off.

This means `1` is true, right? 
I checked the `mstatus.FS` is off, but why it is off. I need to dirty it through some instructions so that those float point instructions will work, do I really need to dirty it every time when I run float point instructions?  Will spike not automatically set `mstatus.FS`? 

Thanks for your help.
Best wishes!

aswaterman, 2024-08-28:
Yes, you need to set FS before attempting to execute any FP instructions.  This is because FS is used for floating-point context control, so the OS can avoid saving/restoring the FP registers.  (It's not about Spike not automatically setting the field.  If Spike automatically set the field, it would ruin the context-control scheme.)

Donkey-Hao, 2024-08-29:
Thanks a lot, I got it.