riscv-software-src/riscv-isa-sim #1778: `iostream` support

mp-17, 2024-08-19:
Does `spike + pk` support C++ `iostream`?

For example, can this program be run with `spike pk`?

```c
#include <iostream>

int main(int argc, char* argv[]) {

  std::cout << "Hello, World!";

  return 0;
}
```

I statically compile it, but during simulation, I get:

`The futex facility returned an unexpected error code.`

Thank you for the support!

muhammadtalhasami, 2024-08-25:
> Does `spike + pk` support C++ `iostream`?
> 
> For example, can this program be run with `spike pk`?
> 
> ```c
> #include <iostream>
> 
> int main(int argc, char* argv[]) {
> 
>   std::cout << "Hello, World!";
> 
>   return 0;
> }
> ```
> 
> I statically compile it, but during simulation, I get:
> 
> `The futex facility returned an unexpected error code.`
> 
> The error appears just by including `iostream`. If I replace the stream with a `printf`, the simulation works ONLY IF I also remove the `#include <iostream>` directive.
> 
> Thank you for the support!

[image]

muhammadtalhasami, 2024-08-25:
@mp-17 for the c++
used this commamd riscv64-unknown-elf-g++ filename.cpp -o filename
and for c use this
riscv64-unknown-elf-gcc filename.c -o filename 
and then simply run it with spike + pk