riscv-software-src/riscv-isa-sim #1795: VI_LD_INDEX issue

EventScheduler, 2024-09-01:
when VLEN=128, the following instructions would load 32 elements in current VI_LD_INDEX implementation:
```
li      t0,76;
vsetvli t1,t0,e16,m4,ta,mu;
vloxei32.v      v4,(s0),v8,v0.t;
```
but it's index width is 32 which is bigger then sew, so I think only "VLEN/32*LMUL = 128/32*4 = 16" elements can be loaded.
In one word, we should also consider the vlmax of indices of vloxei32.v, but not only the vlmax of data. is it right?

DavidYu360, 2024-09-01:
From the code you mentioned, the current vector state (sew, lmul) = (16, 4), and vl = 32 (vlmax).
Since it is a vector index load, it needs 32 index values for 32 data. To describe indexing VPRs, the effective parameters (eew, emul) = (32, 8). The VPR V8 - V15 are for the indexing(emul = 8), and the VPR V4 - V7 are for the load(lmul = 4).

EventScheduler, 2024-09-02:
From data perspective: we will load 32 elements(VLEN/16 * 4 = 32)
BUT from index perspective: we can only load 16 elements(VLEN/32 * 4 = 16).
so we can only load 16 elements finally.
if this is right, then the current VI_LD_INDEX implementation is wrong as it doesn't consider index sew limitation in its for loop.

EventScheduler, 2024-09-02:
my misunderstanding. sorry!

closed 2024-09-02 by EventScheduler