riscv-software-src/riscv-isa-sim #1799: Ssdbltrp - VS mode double trap check

lewislzh, 2024-09-09:
In Spike's double trap check for VS/S mode, the current code accesses `state.nonvirtual_sstatus` in VS mode and `state.sstatus` in S mode.
    [processor.cc#L447-L453](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/processor.cc#L447-L453)
    I think this reverses the intended access to the VS/S status registers, preventing the correct execution of double traps in VS mode. Moreover, I don't think it's necessary to use curr_virt to select between VS and S status registers. 
Spike inherently supports accessing `sstatus` based on the current virtualization mode, reading from either `sstatus` or `vsstatus` as appropriate.
Therefore, we can modify line 466 to: `reg_t s = state.sstatus->read();`

aswaterman, 2024-09-10:
@ved-rivos can you take a look at this?

ved-rivos, 2024-09-13:
> I think this reverses the intended access to the VS/S status registers, preventing the correct execution of double traps in VS mode

Yes. Thanks for spotting that.

> Therefore, we might consider adjusting line 466 to: reg_t s = state.sstatus->read();

I am not sure that is right. Please see use of nonvirtual_sstatus vs. sstatus in rest of the function.

I sent a PR #1804 to swap it. @aswaterman please let me know if its right to just use sstatus->read(). Will it resolve to the sstatus or vstatus internally?

aswaterman, 2024-09-17:
@ved-rivos I'll follow up on the PR.

ved-rivos, 2024-09-18:
@aswaterman - Thanks. I have updated the PR.

PR #1804 (2024-09-18): Fix error in reading right sstatus
closes #1799